Dan-in-CA/SIP #83: Issue with setting up 2 units on LAN

cphgsales, 2015-01-16:
I set up one Opensprinkler pi and when I set up the second I get this error when I try and run the sudo python ospi.py 

plugins loaded:
  mobile_app
  monthly_adj
  relay
  signaling_example
  weather_adj
  weather_level_adj
Starting timing loop

http://0.0.0.0:8080/
Traceback (most recent call last):
  File "ospi.py", line 189, in <module>
    app.run()
  File "ospi.py", line 152, in run
    return web.httpserver.runsimple(func, ('0.0.0.0', port))
  File "/home/pi/OSPi/web/httpserver.py", line 157, in runsimple
    server.start()
  File "/home/pi/OSPi/web/wsgiserver/__init__.py", line 1753, in start
    raise socket.error(msg)
socket.error: No socket could be created

Dan-in-CA, 2015-01-16:
Looks like you need to use a different port number for the second Pi. You can set the port number on the Options page

cphgsales, 2015-01-16:
Where is the options page located?

Dan-in-CA, 2015-01-16:
It can be accessed from the menu buttons at the top of the web interface pages. If you have one unit running you can change the port of that one.

Dan-in-CA, 2015-02-01:
Has this issue been resolved?

cphgsales, 2015-02-02:
Yes it has thank You.

closed 2015-02-02 by Dan-in-CA