Dan-in-CA/SIP #84: Provide a way to config the baseurl, and allow https proxied.

mikedevita, 2015-01-23:
I run nginx at home to proxy all of my web traffic, and drop an ssl in front of everything. 

if you drop an SSL nginx reverse proxy infront of OSPi two defects occur.

1. you are redirected to the http://<hostname>/login instead of https://<hostname>/login
2. some/most of the static assets are loaded via `http://`, when really they should be loaded with `//`, this will allow them to be loaded for both https and http.

I would be happy to fork, and attempt to fix these two things and issue a PR.

Dan-in-CA, 2015-01-23:
that would be great! 
Appreciate the input.

mikedevita, 2015-01-23:
I'm not certain how webpy determines the baseurl to be honest. I got the assets issue fixed already.

Dan-in-CA, 2015-01-23:
Check out:
http://webpy.org/cookbook/ctx

mikedevita, 2015-01-23:
I am making this into a PR instead of an issue. You can delete it if needed.

closed 2015-01-23 by mikedevita