Dan-in-CA/SIP #88: Programs are running for twice the time they were set for

tichris0, 2015-01-26:
Steps:
*Create a new station
**Start Time: 6:00am
**Duration: 60m
**Choose Monday/station
**Select Single Pass
**Save

*View the expected program:
**Hit "Home"
**Go to the next Monday

Note:
The stations is now set to run for double the original time (ie: 2 runs of 60m)

I've been wasting twice the water over this :/.

Dan-in-CA, 2015-01-30:
Are you using the mobile app or the integrated web interface?
I followed your steps but don't see the double time you describe. version 2.2

salbahra, 2015-01-30:
You can also explain this with a watering level of 200%.

tichris0, 2015-01-30:
This is in regards to the built-in page; not the mobile app.

I updated to 2.2.4 since I was on 2.0.1 (RPI SDCard's default) and I don't see that issue anymore.  (It was a bitch to update because copying the old config was generating an "internal error" after logging in.  I had to delete all of my settings for it to work.

On a worst note; I now no longer see any blocks under Mozilla.  I can see that they work under the latest IE, but it's completely broken on Firefox 35.0.1.  The date is not displayed; I can see "Program 1" at the bottom, but no block in the schedule.  Furthermore; clicking next/prev/today has no effect.  I would at least expect it to change the "Program #" at the bottom of the screen (again Firefox only).

Dan-in-CA, 2015-01-30:
Sorry to hear about all the trouble you are having. Updating the program should not require re-doing your settings. The instructions 0n the wiki for updating from an older version should take care of it:
http://rayshobby.net/mediawiki/index.php/Python_Interval_Program_for_OSPi#From_an_earlier_version

The next update will include an updated version of the "system_update" plugin which will allow you to update the software from the UI with just a mouse click.

I have started developing a help system that will be included within the software. For now I think I will add a link to the wiki. However the documentation on the wiki is geared toward installing the software from scratch on a new SD card.

The problems you describe under Firefox are surprising because all my development and testing is done using Firefox. I am running 35.01. It sounds like something in the installation is not right but It's hard to know exactly what.

If it's any consolation, your problems are helping me figure out how to make things easier for the users.
The next major update should be ready within the next week. Hopefully that will clear up your problems.

tichris0, 2015-01-31:
Don't worry I write code for a living so I know how that goes :).  I 
actually did follow the instructions on the wiki and it didn't work.  
I'm guessing that the old settings were miss-read somehow and the code 
went down a bad path causing an 'internal error'. Hopefully the sd-card 
image from opensprinkler gets updated soon so that this won't be an 
issue for anyone else.

The update plugin seems like a good idea.  It will certainly reduce the 
pain of the average user .  I re-loaded the main page today and it 
renders fine in Firefox; so I'm not sure what to make of this. Maybe 
Firefox was in a bad state.  I haven't rebooted the device. I'll let you 
know if I see it again and can gather more data about it.

Thanks,
Chris

Dan-in-CA, 2015-05-01:
Cleaning up old open issues.

closed 2015-05-01 by Dan-in-CA