Dan-in-CA/SIP #89: OpenSprinkler with relay board

eurico31, 2015-02-19:
Hello, 
Can i used OpenSprinkler with a relay board?
I don't  have a OpenSprinkler Pi (OSPi)
what i habe to do for this?

thanks a lot

Dan-in-CA, 2015-02-19:
Yes.
there is a plugin for the program that connects the program to a relay board. But it has not been added to the plugins repository yet because we need to write some instructions for using it.

It should be added in the next day or two. I will post a message here when it is ready.

Do you have the program installed on a Raspi? If you have the latest version of the software installed you can look at how plugins are installed and enabled.

eurico31, 2015-02-19:
thank's
i have Interval Program version : 2.2.27 (2015-02-18)
i wait for update

Dan-in-CA, 2015-02-23:
OK.
The relay_board plugin for OSPi is up. You should see it in the list of plugins to install. Once you have installed and enabled you can click the HELP button for instructions for connecting things up. Or you can fine the instructions at:
https://github.com/KanyonKris/sprinkler-weather/wiki/relay_board-Documentation

eurico31, 2015-02-24:
yes yes !!
thank's lot

closed 2015-02-24 by Dan-in-CA