Dan-in-CA/SIP #93: using modern web technologies like bootstrap

mikedevita, 2015-02-23:
Have you ever thought about using modern client-side web technologies like bootstrap or angular for the sprinkler webui?

It'd clean things up quite a bit and make the code more maintainable. I'd be happy to fork and make an attempt if its something you'd consider merging it in.

Dan-in-CA, 2015-02-23:
I'm certainly willing to consider it.
Do you think it could be done without breaking existing plugins?

When I started this project the idea was just to port Ray's existing Arduino code to Python so it could run on Raspi. I just wanted an irrigation controller to tinker with. Things have kept growing from there.

I chose web.py to implement the UI partly as a tribute to Aaron Shwartz and I wanted to make the software package something that could be downloaded and run without any other set up.

Could you put together a proof of concept prototype?

mikedevita, 2015-02-23:
I certainly could. Worst case I can use it as my own version I suppose. Let me get something started. I'll let you know.

jonathan-marsh, 2015-02-23:
Using such a framework would indeed improve usability (mobile friendly, etc.).

One reason though that I didn't use a framework other than core jQuery (which is pretty much a universal standard at this point) is to limit the number of layers contributors to the project would need to learn to be productive.  Choosing a framework that is minimal in scope and learning curve, and widely understood and adopted, would help maintain that principle.  Would love to know which frameworks you think best qualify!

Dan-in-CA, 2015-02-23:
This is somewhat off topic but this write up in Ars Technina yesterday has a screen shot of the current UI:
http://arstechnica.com/gadgets/2015/02/how-i-upgraded-my-gardens-ugly-drip-system-with-a-sexy-opensprinkler/

Dan-in-CA, 2015-02-23:
Another thing to consider is can we still use the current translation files. So far there is Czech, Spanish and French. The Spanish and French will probably continue to be updated but I'm not sure about Czech.

mikedevita, 2015-02-23:
translation files have nothing really to do with it as you're just changing up how that data is presented I think.

[Bootstrap](http://getbootstrap.com) is a pretty easy curve if you ask me. Its mobile first design, most designers/developers are familiar with it and it has amazing documentation.

Other possibility is  [Zurb Foundation](http://foundation.zurb.com/) which I haven't personally used in production yet, but from what I can tell it offers similar base features. 

Basically what I'd like to see is [bootsrap](http://getbootstrap.com) with [font-awesome](fortawesome.github.io/Font-Awesome/) and jquery. Done right would provide a mobile friendly web-ui thats completely responsive and sexy to boot. Ill see what I can work up in the next few days if I have some downtime.. 

someone else can chime in and get it kickstarted if they like.

martinpihrt, 2015-02-24:
Hi all,
I suppose Rimco refactor version will (when completed) merged into Dan repository? Or Dan and Rimco create each own version (I do not understand)? I contribute to a newer version refactor, I created numerous plug-ins to the newer version - refactor). Of course I will in the future contribute to the Czech language, but into refactor Rimco version (with my plugins, Rimco instalators, logging etc...). I thank you for your reply and clarification of how it actually is (Dan vs Rimco version).
Martin

martinpihrt, 2015-02-24:
actual screen for Rimco revision:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
Thanks Martin

Dan-in-CA, 2015-02-24:
Hi Martin,

Nice to hear from you! 

There are some differences in what the two version of of the project are trying to accomplish. At this point it would be hard to merge Rimco's refactor branch into my original repo.

When Rimco first started his fork he did it in a way that made it difficult for me to contribute to my own project. In fact, the GitHub documentation describes the situation:

```"Historically, the term “fork” has been somewhat negative in context, meaning that someone took an open source project in a different direction, sometimes creating a competing project and splitting the contributors. In GitHub, a “fork” is simply the same project in your own namespace, allowing you to make changes to a project publicly as a way to contribute in a more open manner."```

I have always tried to keep thing simple and accessible so people of every skill level can contribute. It is supposed to be a fun DIY type of project.

There are some features that already exist in my version such as multiple language support, the separate plugins repository, the plugin manager and Blinker signalling which eliminates the need to poll the program for changes. I know Rimco is planning to add some of those to  his version in the future but they probably will done in a different way.

There are also many plugins from several users that have been contributed based on this version.and I have been writing documentation on how to write plugins that would not be valid for his.
https://github.com/Dan-in-CA/ospi_plugins/wiki

Dan

martinpihrt, 2015-02-25:
Hi Dan,
thank you for your response. I'm sorry, but due to a lot of the time that I spent OSPy fork can not imagine that I would plugins rebuilt back to OSPi version. 
Thank you very much Martin

Dan-in-CA, 2015-02-25:
Hi Martin,

I understand. Thank you for your previous contributions.
It is because of  your request for translation support in the UI that we 
now have that feature.

Dan

mikedevita, 2015-02-25:
hey @Dan-in-CA how do you typically do your dev work? Normally on a raspi so you can test sprinkler functions or do you do the web work and the push to a raspi to test? Let me know so I can setup a similar environment for doing the web work.

Dan-in-CA, 2015-02-25:
Hi Mike,
Up till now I have been doing the dev on a Win 7 system. Lately I have been using PyCharm Pro with an open source license. It can send files to the pi but even that is a bit cumbersome. With the new Pi 2 I have started testing a Pi based dev system but so far it has been a bit flaky.

Do you know of a good (free) Linux app for web dev? I have been mostly focused on Python coding and Jonathan Marsh has been designing the UI. I think we are both open to you input.

mikedevita, 2015-02-25:
I dev on a macbook mostly. I use Atom text editor, it looks like theres a linux version [here](https://github.com/atom/atom/blob/master/docs/build-instructions/linux.md). Problem though is how do you actually test code without having access to GPIO? 

**Example:**
Lets say you dev away on a win7 box, codes done now you need to make sure stuff works (i am always bouncing back and forth between browser and developing). Do you push your code and the pull it onto the pi and then visit the URL on the Pi? Thats the only way I know of to test. 

Unless of course like you said you dev on the pi itself.

mikedevita, 2015-02-25:
We should take this offline as its getting off topic. Shoot me an email mike@relative.media.

jonathan-marsh, 2015-02-25:
For dev code (especially on the UI!) I edit in Eclipse and just run Python
locally on my mac, seems to do everything except actually turning valves on
and off.  Just refresh the browser to test new UIs.  I keep an eye on the
data files to ensure stuff is getting set correctly, and upload to the Pi
to test when I’m ready for a more comprehensive test.

-Jonathan

*From:* Mike DeVita [mailto:notifications@github.com]
*Sent:* Wednesday, February 25, 2015 10:21 AM
*To:* Dan-in-CA/OSPi
*Cc:* Jonathan Marsh
*Subject:* Re: [OSPi] using modern web technologies like bootstrap (#93)

I dev on a macbook mostly. I use Atom text editor, it looks like theres a
linux version here
<https://github.com/atom/atom/blob/master/docs/build-instructions/linux.md>.
Problem though is how do you actually test code without having access to
GPIO?

*Example:*
Lets say you dev away on a win7 box, codes done now you need to make sure
stuff works (i am always bouncing back and forth between browser and
developing). Do you push your code and the pull it onto the pi and then
visit the URL on the Pi? Thats the only way I know of to test.

Unless of course like you said you dev on the pi itself.

mikedevita, 2015-02-25:
@jonathan-marsh server actually starts up without being able to talk to the gpio's? Wild. I know when I was dev'ing the nodejs version of a sprinkler app it would barf if it couldn't init the GPIO pins.

jonathan-marsh, 2015-02-25:
I think we put in a bit of code to ensure the thing wouldn’t barf if GPIO
isn’t found, to assist with development scenarios.

*From:* Mike DeVita [mailto:notifications@github.com]
*Sent:* Wednesday, February 25, 2015 12:07 PM
*To:* Dan-in-CA/OSPi
*Cc:* Jonathan Marsh
*Subject:* Re: [OSPi] using modern web technologies like bootstrap (#93)

@jonathan-marsh <https://github.com/jonathan-marsh> server actually starts
up without being able to talk to the gpio's? Wild. I know when I was
dev'ing the nodejs version of a sprinkler app it would barf if it couldn't
init the GPIO pins.

Dan-in-CA, 2015-02-25:
Due to dev on Win7 there are several places in the Python code that get around not having access to the GPIO pins.

william-powell, 2015-04-14:
Running the ospi application on IE 11 has rendering issues. *GASP* I was able to mitigate by adding the following meta tag to base.html

<meta http-equiv="x-ua-compatible" content="IE=edge">

That allowed the app to render properly in IE. That tag should only affect IE. I tested in Chrome, and the web application continued to function.

Thanks for the awesome work on this project!

Dan-in-CA, 2015-04-17:
Thanks William! I have updated the file in the repository.

Dan-in-CA, 2015-04-17:
This issue regarding an update to the UI seems to be inactive Therefore I plan to close it in the next few days.

Dan-in-CA, 2015-05-01:
This seems to no longer be active.

closed 2015-05-01 by Dan-in-CA

Dan-in-CA, 2016-11-03:
Update 11/3/2016:
Bootstrap has been added to SIP and the native web UI works quite well on small screen mobile devices (Phones) with only minor modifications.

mikedevita, 2016-11-03:
thanks @Dan-in-CA ill download the latest and try it out, ive been working on other projects and obv never got time to work on this.

Dan-in-CA, 2016-11-03:
Cool, great to hear from you Mike.

I am still getting to know Bootstrap and have started to try and modify 
the the Timepicker bootstrap plugin. It does not handle setting just 
seconds and minutes out of the box.

At first there was a problem upgrading to the latest jQuery and I 
thought it was going to require a major overhaul of the existing UI but 
it only needed a single line "<!DOCTYPE html>" added to the "base" HTML 
template.

Once jQuery was up to date and I added bootstrap to the "static" 
directory it just started working to my total amazement. I have made a 
few tweaks but there are still some minor problems.

Dan

mikedevita, 2016-11-03:
@Dan-in-CA i loaded it up but it doesn't look any different

```
cd /opt/SIP
sudo git pull origin master
```

Dan-in-CA, 2016-11-03:
It looks the same on a large screen. Do you see bootstrap-3.3.7 in /SIP/static/ ?
If not try "git checkout bootstrap" to switch branches Or try loading it on a phone.

I tried cloning a fresh copy and it is responsive on my phone.