Dan-in-CA/SIP #94: missing gmt offset

dkebler, 2015-02-25:
Just pulled the latest version: 2.2.30 (2015-02-21) and the field for setting the gmt offset is gone.  Now time reads gmt instead of what I had set.

Dan-in-CA, 2015-02-25:
Several IT pros suggested removing that setting. Once the timezone setting in raspi-settings is correct, the time shown in the UI should be correct.

dkebler, 2015-02-25:
not a problem for me.  That's what I did.

But for noobs maybe a problem because now you are making them ssh and run
something from the commandline where before they did not have to ssh to the
pi.   Well maybe noobs are buying the full device in which I assume the web
interface deals with setting time zone.   Alternatively could have a bash
script fired from web interface that sets the timezone instead of using an
offset.  So then..just my, not an IT pro, perspective -:).  You can close
this.

Dan-in-CA, 2015-02-25:
I agree that there should be an easy way for users to do the OS level setup.
At least some detailed instructions in the new help system. I'll work on that for starters.
Thanks for the input.

closed 2015-02-25 by Dan-in-CA