Dan-in-CA/SIP #95: Installed but doesn't run correctly

kenbloch, 2015-02-26:
I installed the software per the directions but I am getting error messages after it starts. 

The http address at the beginning shows as http://0.0.0.0:8080/  (Not as an error)

After the "Starting timing loop" line I get the following:

raceback (most recent call last):
  File "ospi.py", line 204, in <module>
    app.run()
  File "ospi.py", line 157, in run
    return web.httpserver.runsimple(func, ('0.0.0.0', port))
  File "/home/pi/OSPi/web/httpserver.py", line 157, in runsimple
    server.start()
  File "/home/pi/OSPi/web/wsgiserver/__init__.py", line 1753, in start
    raise socket.error(msg)
socket.error: No socket could be created

This is on a new installation of Raspberry PI Noobs.

thep33t, 2015-02-26:
Double check nothing else is running on port 8080 already. Generally that socket error is because it cannot bind to the port.

kenbloch, 2015-02-26:
Brand new installation with one other app running. That app uses port 8443.

Not that it should matter but it is a new Pi 2 model B with 1 GB ram.

Dan-in-CA, 2015-02-27:
I just tried starting a new installation on a Pi 2 with Rasbian 2015-2-16 and it started manually with no problem.

Are you using an install cloned from GitHub or the pre-configured image file?

kenbloch, 2015-02-28:
I am using the cloned from Github. Where can I find the pre-configured
image file? I only have one other app on the Pi at this time so it is not a
problem it I format and start over.

kenbloch, 2015-02-28:
Dan,

Sorry it was another app that caused the issues. It was using the same port 8080 in the background. I solved by starting over and installing the Open Sprinkler first, then changed its tcp port. Then installed other program. Thanks again for you help. ;)

Dan-in-CA, 2015-03-01:
Thanks Ken,
I'm glad it wasn't something more serious.

closed 2015-03-01 by Dan-in-CA