Dan-in-CA/SIP #98: Indexing valve, weather adjustment.

0-kaladin, 2015-03-22:
I have an indexing valve so in order to go through my zones I need to have my pump (zone 1) turn on and off 5 times to tick through my 5 zones.
so i have a start time of 20:00 with duration of 22 min. and then I have recurring every 24 min until 22:00
So when I look at it on the calendar it will tick through 5 times and stop.  

Now I add the wunderground water level and it says it needs to adjust watering 171%.  So it has bumped my 22min to 36 min but not adjusted the stop time so it is still stopping at 22:00, hence only 3 zones get watered.  
Do you know where i would look (which py file) to fix this?  

In the gen2 firmware the UI is a bit cleaner where it just says turn on this zone for X minutes and you can repeat that X times.  In that case even if X minutes was adjusted to longer you'd still repeat X times (in my case always 5 to index through my 5 zones).  I just like the python program better so would rather get this working for me.

Dan-in-CA, 2015-03-22:
It looks like you might need to change line ~283 in helpers.py. Probably to something like:
'''gv.rs[sid][1] = (gv.now + gv.rs[sid][2]) * plugin_adjustment()'''
To understand what the variables are see gv_reference.txt in the OSPi directory.

0-kaladin, 2015-03-22:
didn't seem to effect anything.  I see the values are:
gv.rs	run schedule (list [scheduled start time, scheduled stop time, duration, program index])
but I don't see how they would effect my "until" time for that loop.  
[image]
if you see my program the rain delay is adjust start time and duration and each loops end time but the "Until" is still 2200 so as the duration increased now only 3 loops get done until 2200 where i had it calculated to 5 to loop through my zones.

Dan-in-CA, 2015-03-23:
Oh right. The "until" time is a cut off. anything scheduled to start after that time will not run. I think the weather plugin needs to be changed so that it will only reduce the run time of programs and not increase them depending on conditions. That is, if there is no precipitation the programs will run at 100% of their scheduled time, If there is precipitation the programs should be shortened to compensate for the extra water. I did not write the plugin and have not looked at it closely but I remember seeing something about increasing run times past 100% and thinking that was a bit backwards.

Dan-in-CA, 2015-05-01:
Cleaning up old open issues. The plugins have been moved to:
https://github.com/Dan-in-CA/ospi_plugins

closed 2015-05-01 by Dan-in-CA