Dan-in-CA/SIP #100: no plugins loaded

dkebler, 2015-04-22:
According to wiki any .py files with executable group permission should be loaded.
I have a pull of last master and for example signaling_examples.py and system_update.py  which  I checked are group executable, do not show up in the web interface menu.  

Plugins dropdown shows only "plugins manager" which when clicked does not take one to a manager page (apparently /plugins, according to my page inspection) and "none".

plugin_manager.py is also group executable.

/plugins doesn't make much sense since all other pages are /v?. 

Maybe I am missing some setup detail?   Would like to get weather plugin going but first need any plugin to work.

Dan-in-CA, 2015-04-23:
There was a problem a couple of days ago with the plugins not loading but it was due to the group permissions not being set and has been fixed.
Are you running version 2.2.36? That is running om my test rig and the plugins are loading properly.

If you stop the program with "sudo service ospi stop" then log into the OSPi directory and run 
"sudo python ospi.py" you should see the plugins being loaded or an error message that could help figure out what is going wrong.

dkebler, 2015-05-01:
I did as you suggested and plugins loaded.  DId a reboot.  Plugins loaded.  Now that update plugin is loaded did plugin update to .37 and reboot, all is fine.  Not sure what the issue was but it is resolved.  Thx.

dkebler, 2015-05-01:
FYI  

I noticed when new plugins are loaded via web interface that owner and group are root instead of pi.  Maybe that was/is the issue.  I have chown to pi:pi just to be sure that write privileges work.  Permissions by default seem to be write ONLY for file owner not group.  The plugins directory was always pi:pi.

Dan-in-CA, 2015-05-01:
The problem was caused when I pushed an update to GitHub. The default behavior is for git to unset the permissions.
I changed the code that loads plugins to always load the plugin_manager regardless of how the permissions are set. This should avoid the problem in the future.

closed 2015-05-01 by Dan-in-CA