Dan-in-CA/SIP #102: No initial reset of shift registers

mihugo, 2015-05-26:
I noticed tonight as I was hacking the code to support an old pcduino that I had lying around that there is no initial reset of the shift registers.  If the program should crash while the sprinklers are on and the system gets rebooted via watchdog--or whatever--the sprinklers will stay on until the next automatic cycle.

My suggestion is to add one line to ospi.py to do an initial reset of the shift registers at line 20 right after the first import of set_output. I don't know the code all that well so perhaps there is a more obvious place to stick it.

ospi.py
-    from gpio_pins import set_output
>   set_output()

Dan-in-CA, 2015-05-27:
An excellent suggestion!

There is a bit of a complication however. There is a flag that can be set by a plugin (like the relay_board plugin) when it is loaded that prevents the shift register related GPIO pins from being set up. This allows the plugin to use those pins for other purposes.
if we call 'set_output()' before the plugins are loaded it will cause those pins to be set up and prevent the plugin from working properly.

Let me look into it. There should be an easy solution.

Dan-in-CA, 2015-05-27:
Hmm...
Looks like adding:
    if gv.use_gpio_pins:
        set_output()
at line 234 of ospi.py should do it without problems.

mihugo, 2015-05-27:
Thanks for the quick response. I had a feeling it was too simple to  stick a line of code where I did.  I tried  to confirm your fix worked tonight but ospy.py only has 204  lines of code in the main branch.

Dan-in-CA, 2015-06-01:
Yes, it looks like it was line ~201 for the new code.
I tested the update with the relay_board plugin and there were no problems so I updated the file in the repo. Thanks for the input.

closed 2015-06-01 by Dan-in-CA