Dan-in-CA/SIP #104: Internal server error

kenbloch, 2015-06-08:
I updated OSPI this afternoon and now having two issues. 

First is that when I try to edit programs iI get a message :internal server error" I have deleted programs  and can't create any new etc.

Second error is clock only shows seconds and no minutes or hours.

Dan-in-CA, 2015-06-09:
There were a couple of contributed  updates yesterday. 
The modify.html file in the templates directory was missing a '}' character at column 55, line 284.
The problem with adding programs should be fixed.

I am not seeing the problem with the clock at this time although I did see it briefly just after the update but it cleared on a page refresh.

Update to the latest rev and things should be working.

closed 2015-06-09 by Dan-in-CA

kenbloch, 2015-06-09:
Dan,

That fixed it you can close and delete the issue.

Dan-in-CA, 2015-06-09:
Done. 
Thanks for reporting the problem on GitHub.