Dan-in-CA/SIP #105: Weather level adjust plugin fails on bad location

jking1224, 2015-06-11:
If you enter a bad location in system options, the weather based level adjustment plugin gives an error. 

I am using 2.2.47 OPSi.  I misspelled my town: brommfield. correct is broomfield, co.

Code at line 225 gives: index out of range. I believe the logic flaw is that the wunderground URL call returns a valid json result. But the result is empty.

I'm not a python coder, but I think line 223 should be:

if data is None:
  return ""
if data['RESULTS'] is blank, empty, none (I don't know proper code)
  return ""
lid = "zmw:"......

I tested the URL and the result for a bad query is:
{ "RESULTS": [ ] }

tlo, 2015-06-11:
Hi,
I had the same issue, because my town ("le plessis-robinson" in France) is unknown from the wunderground autocomplete URL. I entered the nearby town to make the plugin work.

Suggestion : could you handle a wunderground ID entered in the "Location" field in the form of "zmw:00000.18.07147" ?

Materdaddy, 2015-06-12:
Weather underground locations do work in the form of: "pws:KCAOCEAN35" which is what I use in the main settings for "location".  I'll look into a pull request for Dan as I've done similar changes to this plugin and jking1224 did most of the investigation already.

What I'm not sure of what you're after here, even if this returns an empty set (your suggestion, and what happens on a failed call where "data" is None) there is no adjustment made.  Are you just looking for a cleaner error in the "Status" field for the plugin?

Dan-in-CA, 2015-06-12:
The fix from Materdaddy has been merged in the plugins repository.
You can re-install the plugin to get the fix.

closed 2015-06-25 by Dan-in-CA