Dan-in-CA/SIP #106: OSPi Installation Wiki - No Repo Labeled "SIP"

Tails86, 2015-06-13:
This is regarding wiki info at https://github.com/Dan-in-CA/OSPi/wiki/Installation. The document refers to https://github.com/Dan-in-CA/SIP. That repo does not seem to exist. I assume it should be https://github.com/Dan-in-CA/OSPi. Or am I missing something?

Dan-in-CA, 2015-06-14:
Sorry for the confusion.

The project is in the process of being re-named from OSPi to SIP. Most of the files now refer to SIP but the repo has not been renamed yet.

ndesimone, 2015-07-26:
Sorry, I know I'm a little late to the game, but FWIW, when I see SIP I don't think of the OpenSprinkler.  I think of VOIP and Session Initiation Protocol.  

I hope the change doesn't lead to confusion, especially for people trying to search the OpenSprinkler and the interval program.

Dan-in-CA, 2015-07-27:
Thanks for the feedback.

This resulted from a lot of considerations especially now that OpenSprinkler has become more commercial. There are also now other options for interfacing a Raspi to irrigation valves such as relay boards (with the relay_board plugin) and the Smart Sprinkler Pi, a 16 port Chinese clone of the OpenSprinkler Pi board.

There are some other meanings for the SIP acronym but I wanted something short (I hate to think of how many times I have typed OSPI) and that hints at water conservation.

Also the new logo took some work and indicates what the project is about:
http://dan-in-ca.github.io/OSPi/

ndesimone, 2015-07-28:
It will take me a while (coming from telco space), but I'll get there.  I totally hear you about typo'ing OSPi, having done that many times myself.

All good.  Thanks for the quick response.

Dan-in-CA, 2015-08-06:
Sounds like I can close this.
If you have any other issues be sure to  open another one.

closed 2015-08-06 by Dan-in-CA