Dan-in-CA/SIP #107: Run Program Manually When Disabled

Tails86, 2015-06-21:
I find it somewhat confusing that the "RUN NOW" button essentially does nothing when the program is disabled. I see under run_now(), the programs page is brought up if the selected program is disabled. 

    if not p[0]:  # if program is disabled
        raise web.seeother('/vp')

Is there is a reason to do that? Personally, I think that there are some that may want to only run programs manually. At the very least, the "RUN NOW" button should be grayed out on a disabled program in the programs page.

Dan-in-CA, 2015-06-25:
Agreed that it would be useful to be able to run disabled programs manually (or from a plugin).
I will modify the code.

Dan-in-CA, 2015-06-29:
I commented out the code that would not allow a disabled program to "run now". The update is in the repo.

closed 2015-06-29 by Dan-in-CA