Dan-in-CA/SIP #108: Mobile app displays many decimals

jking1224, 2015-06-24:
In the mobile app, when a program is running, the mobile app shows many decimals for time remaining:

15:23.99393939991  (and counting down)
I asked the mobile app dev, and he basically stated that the webapp is not sanitizing the values.

tracing this backwards, the mobile_app.py code has cur_settings() /jc  and when I display that in browser (user password hash), it showed:

..."ps": [[0, 0], [0, 0], [0, 0], [1, 831.5999999999999]....
And the code is basically passing back the gv.ps string.  So, in the main program, gv.ps is being filled with too many decimals.

This happens when I run the weather level plugin. The plugin calls for, say 176.4% watering level. When multiplied by program time of, say 15:00, you get a float.

So, in the main program, after any levels (monthly, weather, etc) or plugins have changed the gv.ps array, that array needs to be sanitized back to seconds as integer, no decimals.

Dan-in-CA, 2015-06-25:
I think it should be an easy fix. 
Adding a line of code in the core program to round the value and then convert to an int should work for several plugins.

Dan-in-CA, 2015-06-29:
I pushed a fix to the repo. Let me know if you still see any problem with the number formatting.

closed 2015-06-29 by Dan-in-CA

jking1224, 2015-07-01:
Running 2.2.50, with a weather level factor, lead to a runtime of 15:34 and it displayed as desired with integer seconds, as opposed to seconds with decimals , in the mobile app.  Fixed, thanks

Dan-in-CA, 2015-07-01:
Thanks for letting me know.