Dan-in-CA/SIP #112: Socket Issue After Forcing Exit

Tails86, 2015-06-30:
I have been mostly executing OSPi through SSH. I execute sip.py in a terminal window, do whatever testing I need to do to a plugin, and then close the window once I am done. I know that I really shouldn't be forcing it down like that, so this can be chalked up to user error. I still want to at least note this issue in case someone else is doing something that they shouldn't. I wonder if you could get this same result just by unplugging the system. Occasionally, I run into socket errors when running the program.

    plugins loaded:
      mobile_app
      plugin_manager
    http://0.0.0.0:80/
     Starting timing loop

    Traceback (most recent call last):
      File "sip.py", line 204, in <module>
        app.run()
      File "sip.py", line 157, in run
        return web.httpserver.runsimple(func, ('0.0.0.0', port))
      File "/home/pi/OSPi/web/httpserver.py", line 157, in runsimple
        server.start()
      File "/home/pi/OSPi/web/wsgiserver/__init__.py", line 1753, in start
        raise socket.error(msg)
    socket.error: No socket could be created

It doesn't help if I try to remove data files. This error goes away only when I remove the directory and then clone the repo again. I kept a copy of a corrupted clone if you would like to take a look at it.

Tails86, 2015-07-01:
It seems that rebooting fixes the problem. I swear that I tried it before, but I must have gotten confused.

closed 2015-07-01 by Tails86

Dan-in-CA, 2015-07-01:
This is due to the fact that the built-in server is still running in the background.
If you stop the program with Ctrl C (control key and C) it will avoid the error and you will not need to reboot.