Dan-in-CA/SIP #114: Use a symbolic link for sip.py

ndesimone, 2015-07-26:
I think the instructions are right on the money suggesting that users use the init.d mechanism for starting and stopping OSPi (SIP).  I would suggest that users use a symbolic link rather than moving a copy of sip.py to /etc/init.d.  That way, if improvements are released to sip.py, users will get to take advantage of them right away (especially if they use the upgrade plugin to keep their install fresh).  

I would also suggest putting the "Recommended Method" above the "Detailed Instructions", or maybe moving the "Detailed Instructions" to a "How to Develop" page.  I don't want to make it harder for developers, just easier for the folks that just want to download and get their lawns watered.

I'd make the changes myself, but I'm not sure of the moderation policy/process.  Hope this helps.

Dan-in-CA, 2015-07-27:
I was just looking at the instructions and agree that putting the recommended method first would be best. I will make the change.

closed 2015-07-27 by Dan-in-CA

ndesimone, 2015-07-28:
Hey Dan, what did you think about using a sym-link rather than a copy?