Dan-in-CA/SIP #115: Styling for plugin help pages

ndesimone, 2015-07-28:
Not really an issue, but at some point some thought should be put into the plug-in help pages.  You can see what I mean by clicking the help button in the system update plug in.

I see a couple of options:

- Help appears in the main application window (not my first choice)
- Have the help button open a new "help" window and apply an appropriate styling

Dan-in-CA, 2015-08-06:
The help button does open a new window. this is to allow for easily switching between help and the app pages.

Agreed that a style sheet for the help pages would be a big improvement. I'll see if the style sheet for the UI will work.

Dan-in-CA, 2015-08-06:
added the following line to the plugin docs pages:
    <link href="../../themes/basic/base.css" rel="stylesheet" type="text/css"/>

It is a start. If anyone wants to contribute an improved style sheet for the docs pages it would be welcome.

closed 2015-08-06 by Dan-in-CA