Dan-in-CA/SIP #116: Not running SIP as root

ndesimone, 2015-07-28:
Does anyone know if SIP needs to run as root, or is it safe to have it run as a lower access user?  I'm looking to secure my system and have it connected to the internet so that I can use the mobile client from anywhere.

mihugo, 2015-07-28:
I have been running at a non-root user.  You of course will not be able to run from port 80 (privileged port) without a bunch of extra work.  I am running mine from port 8080 which used to be default

After you change the permission of all the files (including config) files to be the non-root user you will probably want to change the startup script.

My hacks to the init.d are below.  These are just to cause the init script to start the daemon as a different effective user. I think these are all the changes I needed to make it work as a non-root user though somewhere in the back of my mind I recall adding the user to another group but I can't remember why.  Not that it should matter in this case but I'm running mine off an old PCDuino board that I had lying around and hence I have a custom IO libs running on mine that mimic the PI I/O.

I am off two minds for an embedded device if I care if the program runs a root or not.  I could argue on one side that you never want to give root if you don't have to but then I could also argue that for a embedded device that we are just re-arrange deck chairs.

Init.d changes...
Near the top add the LUSER variable and then sprinkle around chuid commands.

..

HOMEDIR=/home/somedirectory/OSPi/
LUSER=somelocaluser

....

do_start()
{
                # Return
                #   0 if daemon has been started
                #   1 if daemon was already running
                #   2 if daemon could not be started
                start-stop-daemon --start --quiet --chdir $HOMEDIR --pidfile $PIDFILE --chuid $LUSER --make-pidfile --background --exec $DAEMON --test > /dev/null
\
                                || return 1
                start-stop-daemon --start --quiet --chdir $HOMEDIR --pidfile $PIDFILE --chuid $LUSER --make-pidfile --background --exec $DAEMON -- \
                                $DAEMON_ARGS \
                                || return 2
                # Add code here, if necessary, that waits for the process to be ready
                # to handle requests from services started subsequently which depend
                # on this one.  As a last resort, sleep for some time.
                RETVAL="$?"

                sleep 1
                chgrp -f $LUSER $PIDFILE
                chmod -f g+w $PIDFILE

                return "$RETVAL"
}

mihugo, 2015-07-30:
Also take a look at Dan's write up on how to setup reverse proxy which is the recommended way to connect to the internet. 
https://github.com/Dan-in-CA/OSPi/wiki/Reverse-proxy

ndesimone, 2015-08-02:
Thanks.  Using a reverse proxy is definitely the way.  Here's the recipe I have been using, http://rayshobby.net/mediawiki/index.php?title=Secure_Remote_Access.  I think one of the important security requirements before opening SIP to the internet was to have fail2ban working.  Within minutes of creating the port map, I started seeing activity.

ndesimone, 2015-08-02:
Reading through both Ray and Dan's versions, I wonder if there is any interest in a harmonized doc (there's good stuff in both that should be used).

Dan-in-CA, 2015-08-02:
Yes, That would be a good idea. 
Both of the existing methods were contributed by separate authors at different  times.

closed 2015-08-06 by Dan-in-CA