Dan-in-CA/SIP #118: SIP vs OSPi and initial configuration

bkoblenz, 2015-08-03:
I am pretty new to this system but it seems you have done some great work and I am looking to make a few extensions.  I have read through most of the code and think I have a fair idea about what is going on.  My first little attempted change is to allow water levels adjustments to be ignored on a per station basis.  But I have a few questions:

1) I understand there are both sip and ospi as primary drivers.  The ospi code seems to enable reverse proxy and thus the sip code seems earlier.  Yet the sip is referenced as the future direction.  On which base should I make changes and why is the reverseproxy not in sip?

2) It seems desirable for a device like this to come up broadcasting some kind of SSID, so a laptop (or phone) could connect to the sprinkler and through a web interface provide the actual SSID and Password that should be used to connect wirelessly to an internal home network.  Presumably once connected, then reboots would automatically try to connect to that network.  I do not see that capability in this package and haven't found a place to leverage.  Any thoughts on this?

Dan-in-CA, 2015-08-04:
Welcome Brian,

I saw your message on my gmail account. SIP will be the new name for the project. I am testing the best way to make the transition without causing too many problems for existing users.

Reverse proxy was a recent contribution from  Chris Bright. I have a SIP version of the project on my local system but it has not been uploaded to GitHub yet.

Broadcasting an SSID can be set up in the OS. There are some instructions on the Adafruit site among others:
https://learn.adafruit.com/pi-wifi-radio/raspberry-pi-setup-1-of-3
look for the section called "Configure wireless network".
Putting the SSID and password in the web interface would be a security problem.

I will send a separate reply to email.

bkoblenz, 2015-08-04:
Thanks for the reply.  I looked at the link, but it basically deals with
configuring linux to have the right ssid and pwd.

That is ok, but I would prefer being able to configure the Pi system to
broadcast an ssid (like OSPi_0xabc) on boot and then when another system
connects to that ssid then they can provide the ssid and pwd that they
really want the Pi system to join.  (Presumably the Pi system would modify
the underlying files referenced in the document and reboot.)

I have wifi thermostats that exhibit this behavior and would suspect it
would be pretty common for IoT devices that need to join a network for
folks who are not too computer literate.  This is the code I am looking to
find and steal!

thanks
-brian

Dan-in-CA, 2015-08-06:
The repo has been renamed and files update. This should resolve any confusion regarding the project's name.

closed 2015-08-06 by Dan-in-CA