Dan-in-CA/SIP #122: Program Individual Stations

cristgal, 2015-08-16:
Dan,

I installed the latest version (Software version: 3.0.12 (2015-08-11)
but I can't figure put how to set a program with specific times for each station, ex:
st1 = 3 min
st2 = 6 min 
st4 = 11 min

is this something that is doable , if yes how?

thanks.

Dan-in-CA, 2015-08-16:
You just need to make a program for each station.
You can schedule them all for the same start time and they will run one after another.

bkoblenz, 2015-08-16:
Go to Programs and click add new program

bkoblenz, 2015-08-16:
Go to programs and click add new program
[image: Inline image 1]

Then specify a starting time, and a duration and the station with the days
of the week you want it to run.  Then click Save.  My station is called S02
and it will run for 15 minutes every day at 6AM.    You can click on the
day of the week and running (or not) that day will toggle.  Also you can
specify multiple stations if you want them to run on the same schedule.
You need a different program for each different schedule.

[image: Inline image 2]

Dan-in-CA, 2015-11-04:
I think this issue has been resolved.

closed 2015-11-04 by Dan-in-CA