Dan-in-CA/SIP #123: Water level adjustment not working

begeles2000, 2015-08-16:
I updated recently to version  3.0.12  and now neither automatic water level adjustment or manual water level adjustments are working. Is this only happening to me?

Dan-in-CA, 2015-08-16:
I just pushed an update that will now show the total water level adjustments including plugins.
I am not seeing the problem you describe. Try updating the software again and if you are still seeing the problem let me know. 

There is now a brand new forum for this project at:
http://nosack.com/sipforum/index.php

If you could post any problem reports there it would be likely to help others as well.

Thanks.

Dan

begeles2000, 2015-08-18:
my bad, the problem was that the water time adjustment is not shown on the schelude for the future, but it is applied when the program is run.

closed 2015-08-18 by begeles2000