Dan-in-CA/SIP #127: Will not start after update

mshill, 2015-08-22:
I'm using SIP on beaglebone black, and after the latest update, I can't seem to start SIP. After update I got "server error" and then trying to start at the command line, I get:
=============================
Traceback (most recent call last):
  File "sip.py", line 16, in <module>
    import gv
  File "/home/ubuntu/SIP/gv.py", line 46, in <module>
    from helpers import password_salt, password_hash, load_programs, station_names
  File "/home/ubuntu/SIP/helpers.py", line 26, in <module>
    from gpio_pins import GPIO, pin_rain_sense
  File "/home/ubuntu/SIP/gpio_pins.py", line 61, in <module>
    pin_rain_sense = gv.pin_map[15]
IndexError: list index out of range
=============================
I live in the HOT desert and downtime on the sprinkler system isn't good... :(

Dan-in-CA, 2015-08-22:
For now you can roll back to an earlier version.

From the command line cd to the SIP directory and issue the command:
<pre>sudo git checkout 4bc3f05f</pre>

That should get you going until we can fix the problem.

mshill, 2015-08-22:
Excellent, THANKS!

Dan-in-CA, 2015-08-22:
I just pushed an update that should fix the problem.

DanielCasner, 2015-08-23:
This may be the same issue or may be separate but with the latest software (on a Raspberry Pi B with the first gen OSPI hat) after doing a soft resert (such as after installing pluggins) valves can not be controlled. There is a warning about "unrecognized service" but other than that SIP reports it's working fine, it just doesn't turn on any valves. Killing python and restarting fixes the problem.

Dan-in-CA, 2015-08-23:
The warning about "unrecognized service" indicates that the set-up is not using the /etc/init.d script which is required for the auto-restart to work.

It you are using rc.local to auto-start the software it is recommended to switch to what is now the recommended method as described in the latest setup docs:
https://github.com/Dan-in-CA/SIP/wiki/Installation
Scroll down to "Starting the Program Automatically.

I seem to recall that you forked the OSPi repo quite a while back. You may be one of the longest users.

Thanks.

DanielCasner, 2015-08-26:
Thanks Dan. The problem was mine for being lazy about applying the rename. Thanks for all your work on this project. I did fork it a long time ago but now that the plug-in architecture is there, it's must less nessisary to have a fork. Hopefully I'll have time to clean up my features and submit them as plug-ins soon.

Dan-in-CA, 2015-08-26:
Cool! 
New features are always welcome.

closed 2015-08-26 by Dan-in-CA