Dan-in-CA/SIP #129: response time webserver ..

aocana, 2015-09-04:
Hi!

I have two issues with SIP software. First of one is related with response time of webservice, I have been checking the times of responses and is very high, you can see in the following command returns lookup, connect, pretransfer, starttransfer time in seconds and the total time that the full operation lasted:

~  curl -s -w '\nLookup time:\t%{time_namelookup}\nConnect time:\t%{time_connect}\nPreXfer time:\t%{time_pretransfer}\nStartXfer time:\t%{time_starttransfer}\n\nTotal time:\t%{time_total}\n' -o /dev/null http://raspi-chapitown/

Lookup time:	0,005
Connect time:	0,009
PreXfer time:	0,009
StartXfer time:	25,870

Total time:	25,880

---- end log --

The second issue is rated with the programs execution. All of them works fine least one, and to do it work I must stop the service and re-starting the service again. The software looks that the program-1 works but it not sends the command to pump can start , do you know why to start this  program-1 I must to restart the service manually from console raspberry-pi with command "sudo service sip stop" and "sudo service sip start"

Thanks for your comments!

I look your support!

Dan-in-CA, 2015-09-04:
Hi,
First issue has been a known problem problem. I am looking for a solution.

Second issue:
I will need more information abut your setup to find the problem.
I am guessing you have the pump on a station that is set as Master and the station that runs the water valve is set to Activate master on the Stations page.

Are you using a relay board or the OpanSprinkler board?
What version of the software are you using?
Is there any thing else about program-1 that is different from the other programs?
Do other programs/stations use the pump?

FYI:
You can also restart the software from Options>System Restart>Restart

aocana, 2015-09-05:
hi !! thanks for your soon answer. There is any way to redirect output log of service sip that start at run to a file to see what happens?
because the problem only happens if I reboot the system . After stop the service and start again everything works fine!

aocana, 2015-09-06:
The setup of system configured is all station in line I mean do not have a master pump.
I'm using relay board connected to my raspberry pi and the version installed of SIP is: 3.1.25 (2015-09-03).There are not differences between the progam-1 and other programs except the time that is scheduled the execution.
I'm attaching a log output of console when I have the issue with program-1 failing (manual execution porgram-1 for one minute):

--- start log ---

pi@raspberrypi-:~/sip$ sudo sip.sh
sudo: sip.sh: command not found
pi@raspberrypi-:~/sip$ sudo python sip.py
Ignoring exception while loading the relay plug-in.
cannot import name pin_relay
Complementos cargados:
  email_adj
  mobile_app
  monthly_adj
  plugin_manager
  proto
  relay_board
  signaling_examples
  system_update
  weather_adj
  weather_level_adj
Comenzando tiempo de blucle

zones changed
[0, 0, 0, 0, 0, 0, 0, 0]
http://0.0.0.0:80/
Monthly Adjust: Setting water level to 90%
Email plugin is started
Checking weather status...
Checking rain status...
Good weather detected: Partly Cloudy. Removing rain delay.
Skipped wundergound data because of a parsing error for today
Using 7 days of information.
Water needed (7 days): 21.0mm
Total rainfall       : 4.0mm
_______________________________-
Irrigation needed    : 17.0mm
Weather Adjustment   : 60.7%
192.168.1.128:57726 - - [06/Sep/2015 13:12:21] "HTTP/1.1 GET /" - 200 OK
192.168.1.128:57728 - - [06/Sep/2015 13:12:21] "HTTP/1.1 GET /" - 200 OK
192.168.1.128:57726 - - [06/Sep/2015 13:12:24] "HTTP/1.1 GET /static/themes/basic/base.css" - 200
192.168.1.128:57900 - - [06/Sep/2015 13:12:24] "HTTP/1.1 GET /static/scripts/jquery-1.8.2.min.js" - 200
192.168.1.128:57901 - - [06/Sep/2015 13:12:24] "HTTP/1.1 GET /static/scripts/behaviors.js" - 200
192.168.1.128:57902 - - [06/Sep/2015 13:12:24] "HTTP/1.1 GET /static/scripts/schedule.js" - 200
192.168.1.128:57903 - - [06/Sep/2015 13:12:24] "HTTP/1.1 GET /static/scripts/intervalSelect.js" - 200
192.168.1.128:57900 - - [06/Sep/2015 13:12:27] "HTTP/1.1 GET /static/themes/basic/logo.png" - 200
192.168.1.128:57726 - - [06/Sep/2015 13:12:27] "HTTP/1.1 GET /static/themes/basic/grass.jpg" - 200
192.168.1.128:57901 - - [06/Sep/2015 13:12:27] "HTTP/1.1 GET /static/themes/basic/on.png" - 200 192.168.1.128:57902 - - [06/Sep/2015 13:12:27] "HTTP/1.1 GET /static/themes/basic/home.png" - 200

Traceback (most recent call last):
  File "/home/pi/sip/web/application.py", line 239, in process
    return self.handle()
  File "/home/pi/sip/web/application.py", line 229, in handle
    fn, args = self._match(self.mapping, web.ctx.path)
  File "/home/pi/sip/web/application.py", line 438, in _match
    what, result = utils.re_subm('^' + pat + '$', what, value)
  File "/home/pi/sip/web/utils.py", line 531, in re_subm
    compiled_pat = re_compile(pat)
  File "/home/pi/sip/web/utils.py", line 499, in __call__
    update(block=True)
  File "/home/pi/sip/web/utils.py", line 496, in update
    self.running[key].release()
error: release unlocked lock

192.168.1.128:57904 - - [06/Sep/2015 13:12:28] "HTTP/1.1 GET /api/status" - 500 Internal Server Error
192.168.1.128:57903 - - [06/Sep/2015 13:12:29] "HTTP/1.1 GET /api/log" - 200 OK
192.168.1.128:57900 - - [06/Sep/2015 13:12:30] "HTTP/1.1 GET /static/themes/basic/whitestripes.png" - 200
192.168.1.128:57919 - - [06/Sep/2015 13:12:51] "HTTP/1.1 GET /vr" - 200 OK
192.168.1.128:57919 - - [06/Sep/2015 13:12:52] "HTTP/1.1 GET /api/log" - 200 OK
[... 114 lines cut ...]
192.168.1.128:58081 - - [06/Sep/2015 13:29:40] "HTTP/1.1 GET /api/status" - 200 OK
192.168.1.128:58081 - - [06/Sep/2015 13:29:41] "HTTP/1.1 GET /api/status" - 200 OK
192.168.1.128:58081 - - [06/Sep/2015 13:29:42] "HTTP/1.1 GET /api/status" - 200 OK
192.168.1.128:58081 - - [06/Sep/2015 13:29:43] "HTTP/1.1 GET /api/status" - 200 OK
192.168.1.128:58081 - - [06/Sep/2015 13:29:45] "HTTP/1.1 GET /api/status" - 200 OK
192.168.1.128:58081 - - [06/Sep/2015 13:29:46] "HTTP/1.1 GET /api/status" - 200 OK
192.168.1.128:58081 - - [06/Sep/2015 13:29:47] "HTTP/1.1 GET /api/status" - 200 OK
192.168.1.128:58081 - - [06/Sep/2015 13:29:48] "HTTP/1.1 GET /api/status" - 200 OK
192.168.1.128:58081 - - [06/Sep/2015 13:29:50] "HTTP/1.1 GET /api/status" - 200 OK
192.168.1.128:58081 - - [06/Sep/2015 13:29:51] "HTTP/1.1 GET /api/status" - 200 OK
192.168.1.128:58081 - - [06/Sep/2015 13:29:52] "HTTP/1.1 GET /api/status" - 200 OK
192.168.1.128:58081 - - [06/Sep/2015 13:29:53] "HTTP/1.1 GET /api/status" - 200 OK
192.168.1.128:58081 - - [06/Sep/2015 13:29:54] "HTTP/1.1 GET /api/status" - 200 OK
zones changed
[0, 0, 0, 0, 0, 0, 0, 0]
zones changed
[0, 0, 0, 0, 0, 0, 0, 0]
192.168.1.128:58081 - - [06/Sep/2015 13:29:56] "HTTP/1.1 GET /api/status" - 200 OK
192.168.1.128:58081 - - [06/Sep/2015 13:29:56] "HTTP/1.1 GET /api/log" - 200 OK

--end log --

is there any way to see the output of script "sudo service sip stop" ?

thanks for your help!!

aocana, 2015-09-06:
we continue this thread in next link:
http://nosack.com/sipforum/showthread.php?tid=10&pid=29#pid29

Dan-in-CA, 2015-09-11:
OK I will close this issue here.

closed 2015-09-11 by Dan-in-CA