Dan-in-CA/SIP #130: Relay plugin will not load

jking1224, 2015-09-05:
running version 3.1.25, I previously had relay plugin running. Now it does not appear in list of installed plugins (the menu being the list).
I stopped the sip service and ran sudo python sip.py

got message:
ignoring exception loading plugin relay
cannot import name pin_relay

jking1224, 2015-09-05:
Doing some digging, it seems gpio_pins.py was changed substantially, removing the global variable pin_relay.

Dan-in-CA, 2015-09-05:
Yes, there have been some major changes. A new contributor has been making some really good improvements but there have been a couple of bugs.
I am working on this.
So far I have the plugin loading but it does not activate the relay on my test rig.

bkoblenz, 2015-09-06:
Sorry to all.  This is probably my error.

I may have deleted that variable by mistake.

I dont see any uses except in relay_board.py and there only a definition.

I have been using the relay_board plugin and it works for me (I am using a
single relay board and not OSPi version.)
I can turn relays on and off, but they are directly connected to IOs.
Also, I am using a Pi2, not earlier version and not Bo.

Dan, is there something more you want me to look at?  Again, my apologies,
but I dont yet see what I did that would cause the problem.

-brian

Dan,

jking1224, 2015-09-06:
some versions of the Raspberry Pi openspinrkler board have a relay mounted on them. The "test relay" plugin would trigger the relay on/off.  Yes, we need that variable restored in the code and its most recent set up/initialize behavior put back.

bkoblenz, 2015-09-06:
I think I made the change to put this back.  Dan, see my note in pull
request.

sorry.

Dan-in-CA, 2015-09-06:
I tested the update but it does not activate the relay on my test rig. Testing an older version of the software does activate the relay.
I will do some further work on this.

Dan-in-CA, 2015-09-06:
@jking1224 
As a temporary fix you can roll back to an earlier version.
from the SIP directory:
<pre> sudo git chechkout 4bc3f05fd8</pre>

This works here.

Dan-in-CA, 2015-09-06:
The strange thing abut the relay pin not working is that pin_rain_sense which has similar mapping does work but pin_relay does not.

Dan-in-CA, 2015-09-06:
OK. I just pushed an update which I think fixes the problem.
I was using pigpio and that made it not work on my test system. I will be pushing an updates relay plugin also that works with both GPIO libraries.

bkoblenz, 2015-09-06:
Thanks. I guess I'll see what I messed up. 

What do you think about over time capturing log in file?

The log level can be adjusted as a parameter to sip.py

Dan-in-CA, 2015-09-06:
No problem. These things happen. I've actually introduced a bug once myself.
The way I tracked down the problem(s) was to modify the Except block in relay.py to:
<pre>        except Exception, e:
            print "Relay plugin error: ", e </pre>
This outputs the exception message from Python.

Yes. error logging would be good for debugging As long as we can turn it off when not needed. This helps keep the SD card from failing due to it's finite number of write cycles.

bkoblenz, 2015-09-06:
I am not quite getting what was wrong.  What change did you make?  (For
some reason merging with your sip.py makes everything look different, so
that is not too helpful.)

It looks like possibly an extra else: clause in gpio_pins.py but I do not
have that in my code so not sure if I am seeing that right.

-brian

Dan-in-CA, 2015-09-11:
as far as I know this has been resolved. 
If anyone is still having this problem, please open another issue.

closed 2015-09-11 by Dan-in-CA