Dan-in-CA/SIP #132: Programs not running

ndesimone, 2015-09-10:
One of the updates I've done (sorry, don't know which) has broken my programs from running.  Basically, they don't fire at the preset time.  One thing I noticed is that I get "internal server error" when navigating to the "Stations" page.  I'm not sure if those are related.

To get things running enough to run a program manually takes a full reboot.

- How should I debug this?  Is there a log somewhere?
- How do I figure out why there is a server error?  I guess this is the same question as above, is there a log I can check?

Thanks,
Nick

bkoblenz, 2015-09-10:
Unfortunately, there are no logs for this.

The easiest way to debug this is to stop the sip.py program from running as
a daemon:
$ sudo /etc/init.d/sip stop

Then go to the directory where sip.py is and run it:
$ sudo python sip.py

You will get some output.  Then when you go to the webpage and click the
stations tab, there will be an exception that shows up in the window that
is running sip.py.  That should hopefully lead to a line number where the
problem was.  (Probably somewhere processing templates/stations.html.

As Dan would say, make sure you have the latest and greatest of everything
first!
-brian

ndesimone, 2015-09-10:
Aha!  Not sure the cause yet, but at least I have more to go on...

  File "templates/stations.html", line 120, in __template__
    <input type="checkbox" ${"checked" if gv.sd['mo'][bid]&(1<<s) else ""} id="mc${sid}">
IndexError: list index out of range

192.168.123.33:62321 - - [09/Sep/2015 19:36:15] "HTTP/1.1 GET /vs" - 500 Internal Server Error

ndesimone, 2015-09-10:
I think this was caused by sd.json.  I nuked the files in data so that SIP would rebuild them.  The value in the "iw" is [0, 0, 0].  In the original file, the value for "iw" was [0].  Seems like that could lead to a index out of range error.  

Did I miss a step on an upgrade?

bkoblenz, 2015-09-10:
Sounds like you are definitely on to something.

iw is indexed by the number of extension boards you have which is
determined by your station count.  Did you have an iw with only one entry
and more than 0 extension boards (more than 8 stations)?

How many boards do you have?

Also, did you notice the 'mo' array?  It too is indexed by board number and
if you have a master valve might have the same problem.

-brian

ndesimone, 2015-09-10:
I have an expansion board which counts as two.  This means that the array *should* have 3 entries.  I don't have a master valve.  It looks like 'mo' was correct ("mo": [0, 0, 0]).

After the rebuild, SIP is working correctly again.  Yay!

A couple of things:
- This has happened before to me during upgrade (that's why I knew to look for it).  Is there a way to protect against this?
-- An upgrade strategy (like rails has for database migration).  This forces us to explicitly plan for data changes.
-- High level export/input.  This could be a simple recovery mechanism or part of the upgrade, i.e. export config, update, import config.
- A logging facility to find errors after the fact
-- If we redirect SIP output to /var/log/sip.log and build a logrotated rule for it, it could help us find problems faster.

bkoblenz, 2015-09-10:
These are good ideas.  Want to implement them:-)

MrWendall, 2015-09-11:
Greetings,
Same issue with programs not running after update. My SprinklerPi is on other side of the planet just to make things more fun.

Fortunately I kept screen shots of all my settings so I have been able to recreate the programs.

The pluggin that caused the issue for me was the one that helped locate valves by making them chatter. I think it was called blink. I removed it and updated again from the web interface.

I can now see the programs appearing properly on the home page of the web interface but they are not starting. I have been at least been able to get the run once program running so can stop everything from dying.

**Could some please advise on how to rebuild SIP?**
Below is the output from 

sudo python sip.py
plugins loaded:
  mobile_app
  plugin_manager
  signaling_examples
  system_update
zones changed
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
Starting timing loop

http://0.0.0.0:8080/
10.8.0.4:29901 - - [11/Sep/2015 12:14:47] "HTTP/1.1 GET /login" - 200 OK
10.8.0.4:29901 - - [11/Sep/2015 12:14:52] "HTTP/1.1 GET /api/log" - 200 OK
someone logged in
10.8.0.4:29901 - - [11/Sep/2015 12:14:55] "HTTP/1.1 POST /login" - 303 See Other
10.8.0.4:29901 - - [11/Sep/2015 12:14:58] "HTTP/1.1 GET /" - 200 OK
10.8.0.4:29901 - - [11/Sep/2015 12:15:03] "HTTP/1.1 GET /api/log" - 200 OK
10.8.0.4:29916 - - [11/Sep/2015 12:15:04] "HTTP/1.1 GET /api/status" - 200 OK
10.8.0.4:29901 - - [11/Sep/2015 12:15:04] "HTTP/1.1 GET /api/log" - 200 OK
10.8.0.4:29901 - - [11/Sep/2015 12:15:08] "HTTP/1.1 GET /vs" - 200 OK
10.8.0.4:29901 - - [11/Sep/2015 12:15:14] "HTTP/1.1 GET /api/log" - 200 OK

Dan-in-CA, 2015-09-11:
MrWendall.
I see you have an expansion board by the number of stations shown. Try setting your system to 0 expansion boards on the Options page and then setting back to the number of boards you actually have. This may fix the problem.

Let us know if it works or not.

MrWendall, 2015-09-12:
Hi Dan,
Thanks for your prompt response.
Unfortunately changing the number of extension boards, has had no effect.

I am sorry if a bit thick but I cant find instructions for "rebuilding" SIP as mentioned above.

Thanks

Dan-in-CA, 2015-09-12:
When I tried re-setting the number of extension boards on my test system it updated the "iw": [0] element in the SIP/data/sd.json file to "iw": [0, 0, 0],. This is what Nick indicated solved the problem (He had the equivalent of 2 expansion boards). It was necessary to click the "Submit Cnanges" button at the bottom of the Options page after each change in the number of extension boards.

What Nick was referring to when he said "rebuild" was to delete (or rename) the sd.json file in the SIP/data/ directory. You could do this from the SIP directory with
<pre>sudo mv data/sd.json /data/sd.bak</pre>
This will rename the file and when you restart the program it will be automatically re-generated. It will re-set any settings you have to defaults and you will need to adjust any you have changed. 
You can restart the program quickly with:
<pre>sudo service sip restart</pre>

MrWendall, 2015-09-12:
You are a Legend.

Will give it a shot and report back.

Thanks again

Dan-in-CA, 2015-09-12:
If all else fails there is a way to roll back to an earlier version. You would need the original (backup) copy of the sd.json file. Let me know how it goes.

MrWendall, 2015-09-12:
G'day Dan,
No joy.

Have winSCP into SIP to try and edit the iw parameter in the sd.json to no effect.

Happy to reprogram SIP but no physical access to machine and kind of stuck on how to proceed.

Thanks again.

Dan-in-CA, 2015-09-12:
I would suggest the rollback as described in the edited version of the previous comment.
If you have ssh access to the system use (from the SIP directory)
<pre>sudo git checkout 4bc3f05f</pre>

Then restart:
<pre>sudo service sip restart</pre>

That should get programs working.

MrWendall, 2015-09-12:
You are a good man Dan. Thank you. 
Lesson learned about remotely updating firmware.
Fingers crossed that it stays working.
Cheers

Dan-in-CA, 2015-09-12:
Now I will be able to sleep tonight.

MrWendall, 2015-09-12:
Seriously, Thank You!

You have save our vegetable and herb gardens but most importantly some orchids that would be almost impossible to replace.
Can you private message me your PayPal details so I can buy you a drink?

ndesimone, 2015-09-12:
Sorry I couldn't join all the fun (high school football game).  Thankfully, it sounds like you guys got it sorted. 

I only ever access my OpenSprinkler via remote so I think you'll be ok as long as you're not messing with OS stuff.  Dan suggestions are great for getting it working quickly.

The next time you want to try the upgrade, I would take screen shots of all your programs and station definitions as well as copy off the .json files in the data directory.  If you run into problems:

- Stop the software (sudo service sip stop)
- Remove all the .json files in the data directory
- Restart the software (sudo service sip start)
- Manually reenter the programs from the UI to rebuild the .json files.

We should be able to diff the two sets of files and figure out what happened.  I'm not sure what timezone you're in, but, if we can prearrange a date/time, I can jump on a Skype call with you.

Dan-in-CA, 2015-09-12:
@ MrWendall,
I don't have a PayPal account (yet) but there is a new SIP forum:
http://nosack.com/sipforum/index.php
When you get a change maybe you could post some pics of your orchids. That would be thanks enough.

closed 2015-11-04 by Dan-in-CA