Dan-in-CA/SIP #133: sip.sh should be executable

ndesimone, 2015-09-10:
Is there a way to make sure that sip.sh stays executable?  

I have /etc/init.d/sip sym linked to /home/pi/SIP/sip.sh so that I'm always using the latest and greatest.  Last time I ran the upgrade program, it overwrote sip.sh (expected and desired), but it set the mode to 644 (not desired).

Thanks,
Nick

ndesimone, 2015-09-10:
I just did an update to take the last published update (one line change in sip.py).  It still changed the file perms on sip.sh which makes it so SIP can't restart on boot.

Dan-in-CA, 2015-09-11:
Sorry for the delayed response. Been out of town for the day.
Part of the problem is that a lot of the development is done on Windows systems. 
On the Linux side you can set Git to pay attention to file permissions:
sudo git config core.filemode true

For this reason I try to push updates from my Pi test rig. In this case I didn't know that anyone was symlinking to sip.sh in the SIP dir. I just pushed a copy of sip.sh with permissions set to exicutable. That file doesn't usually change with updates so it should maintain those permissions.

ndesimone, 2015-09-11:
No worries.  THANKS!

ndesimone, 2015-09-11:
Just did a test.  That worked perfectly.  Thanks Dan!

closed 2015-09-11 by ndesimone