Dan-in-CA/SIP #135: water hammer reduction

greglarious, 2015-09-25:
this is more of a helpful tip suggested for the wiki rather than a code change:

my current sprinkler system (and all of the sprinkler systems i've owned) makes a big thunk sound in the house pipes every time the zone changes.  by setting the station delay to -1 second, you end up scheduling the opening of the new zone solenoid to occur just before closing the previous zone.  this eliminates the shock to the water flow between zones.  using this technique, you can reduce the number of pipe thunks to just one at the end of the program when the last zone shuts off.

the only down-side of this technique is briefly operating 3 solenoids (master, current zone, next zone) before going back to 2.  seems harmless enough unless anyone knows why this would be a problem.

andrewradke, 2015-09-26:
Many solenoids are "slow close" to avoid exactly this problem. Where you have problems with water hammer I would suggest that using these would be better.

The ones I have like this are the same as the ones we use at work and are generally used in agriculture and horticulture (at least here in Australia).

greglarious, 2015-09-26:
perhaps a plumbing hardware solution would be a better engineered solution, but the valves i have are only 3 years old and were the best the local hardware store provided at the time.  using the 1 second overlap i am able to reduce hammer effect from 8 events to 1 event per cycle so this adds a good bit of value with no plumbing work.  what i should do is use the 1 second overlap technique on the first 7 zones and replace just the final solenoid with a slow close model to eliminate the final hammer.

Dan-in-CA, 2015-09-26:
There are also water hammer arresters that you can get for less than $20.00. Check with your hardware store.

closed 2015-11-04 by Dan-in-CA