Dan-in-CA/SIP #138: mobile/responsive front-end

dkebler, 2015-10-27:
Has anyone begun an alternative or additional front-end that is responsive and thus usable in a mobile device? 

I code in javascript/node and my first thought was a complete rewrite but as I looked into what as been done a new front-end to the existing python code seems way easier. 

Maybe rather than recreate the entire existing front-end I also thought that just a simple subset of the features/settings would be enough so folks can operate their system from their phone.   Any significant setup could still be done with the existing interface on a laptop/desktop.

Thoughts anyone?   Some discussion about the details of how currently the web interface accesses the backend functionality would be helpful (api?)

Dan-in-CA, 2015-10-27:
Hi David,

As you may know Samer Albahra has written a mobile front end app that has worked with both OpenSprinkler and OSPi (now SIP).
https://github.com/OpenSprinkler/OpenSprinkler-App

As far as I know he may be planning to drop support for SIP. Another contributor to the project has offered to contribute to Samer's project to maintain support for SIP but I am not sure how that is progressing. One of the problems Samer cited with his support of SIP is the fact that most of the new functionality is added through plugins and that is not something that can be supported with OpenSprinkler.

Back in February a user offered to develop a new UI using bootstrap.
https://github.com/Dan-in-CA/SIP/issues/93
For some reason he dropped off the map. I tried to contact him via direct email and got no response.

Your idea of developing a minimal front end for mobile use sounds very interesting.

The current web interface is built using Web.py.
http://webpy.org/
There is a beginning of a REST API but we have been using a plugin to provide an API for Samer's app up to this point.

I have also started experimenting with a whole new way of interfacing a Python program with JavaScript. There is a preliminary discussion on my blog:
http://xperimentia.com/2015/08/20/building-a-browser-based-user-interface-with-mqtt/
Things have developed quite a bit since I posted that and I now have a timer system using MQTT to interact with a Bootstrap UI using direct messaging between Python and JavaScript. I'm not sure if it is worth re-doing the entire SIP UI but there are some interesting possibilities for adding functionality.

Dan

dkebler, 2015-10-29:
Given your reply....more thoughts for food.

A public REST-API is ultimately the best way to go (? or websockets maybe) in terms of liberating UI coding from all the internals.   I have another project I am starting which controls a relay bank for lighting.  If I write a browser responsive UI for that then it's not a stretch to adapt it to another relay device as long as it has a REST-API.   

Thinking a KISS baby step  ---  If I am a user mostly I want is manually turn on/off a station/relay while I am out in the yard.  How about starting with just that functionally.  What actions would be required to do just that?

??
1.  Authenticate User
2.  Display a list/table of stations with buttons and current state.
3.  Listen for a user change of relay state request.
4.  Fire relay change-state code.

Looks like maybe three API functions??,  authenticate, read a requested table (relays), Toggle state(s) of requested relay(s).

Not sure at all how easy it is to in python to code a REST-API but I assume similar packages exist to those in node.js  http://python-eve.org/.  I guess you even have a start on that?  That would be on you as I don't at this time want to take my coding focus away from node.js.   

as to MQTT Node.js running on Wheezy Debian is an issue but Raspian is now available as Jessie which supports node.js install.  I have yet to get that set up on an RPi but that is a priority project as I want to be able to code in node.js on the RPi.   So given that one could write modules in js that would then talk to your python code using MQTT through a private "message "interface.  That would allow the possibility that the REST-API and http-server could be node based.  One could slowly if they want rewrite parts of the python code in js or write new code in js, all the while the existing UI will work fine.

A few other things to consider would be not using monolithic (maybe overkill maybe antiquated) "frameworks" like bootstrap and foundation.  One can now with the help of things like gulp put together pieces that provide only the needed functionality and don't lock anyone into something in particular, like using less with bootstrap.   Of course with a REST-API folks can do what they please when it comes to the UI (and I will).

Me personally I'm opting for a material design based UI, but I have a gulp workflow to integrate npm packages and bower libraries as I see fit.  

Also of interest if one wanted to write a stand alone android app (not really necessary imo) or even a stand alone desktop app there are groovy things like this now.  http://nodeappkit.owinjs.org/    http://nwjs.io/   http://electron.atom.io/  which simply leverage your web UI code.

---
BTW
The relay board I have currently has a socket API for control so maybe I'll opt for websockets as then that means the browser could talk directly to the relay board once a configuration file is loaded(cached) from a server somewhere.

dkebler, 2015-10-31:
came across this today https://www.rabbitmq.com/  Looks like another way the existing python code could message with javascript.  It is not as "lightweight" as MQTT probably.

Dan-in-CA, 2015-10-31:
Yes, I saw this video about RabitMQ a couple of weeks ago:
http://mashable.com/2015/10/07/what-is-rabbitmq/#XS7LNh5jC5qJ
looks like it is geared toward cloud infrastructure development.

I am now using the mosquitto MQTT broker (server) to also serve HTML an JS files. It is written in C so it is fast and light weight.  It interfaces with JavaScript through websockets which requires building from scratch to enable websockets. Mosquitto also includes a lot of security features..

Qry3vavna, 2015-11-01:
RabbitMQ is a beast as far as routing and capabilities go. Specifically designed by the banks and high tech companies for secure guaranteed delivery. It might be overkill for this project, though if it was running on a different more powerful box than a BBB, there are plug-ins for Rabbit to talk almost any other middle-ware protocol including MQTT. MQTT is lighter weight and used more in embedded systems.

Also, thanks Dan for writing this!
I'm using the BBB with OsPI in my back yard, with a few zones setup for raised beds and included a rain sensor.  Plans for more will have to wait until I get more land. :)

dkebler, 2015-11-04:
I successfully put raspbian Jessie, node 5 and npm 3.3.11 on a SD card.  Runs fine.  Even ran a little JavaScript chat server socket.io app on it.  Performance was fine , works great.   So yes it seems one can now program in node on a RPi using the latest node.

Then I decided to load some opensprinler firmware which lead me to checking out the the ospi firm/software again and it looks much improved in particular it has now the one feature that was missing namely the weather underground tie in which was what had me using SIP.   Their UI is already responsive so DRY wise I'm on to other projects, sorry.

Dan-in-CA, 2015-11-04:
No problem. Thanks for stopping by.

This project is for those who like to tinker and customize both software and hardware.

Dan

closed 2015-11-04 by Dan-in-CA