Dan-in-CA/SIP #139: System update versions

ndesimone, 2015-11-08:
I'm currently running version 3.1.34 (revision 34).  The system updater is reporting that there is an upgrade to revision 28.  

Currently running revision: 34 (2015-10-26)
Available revision: 28 (2015-11-03)

I looked in the repo, and there are more recent changes in the code line (Italian translation), and they match the dates above.

Did something happen with the versioning process?

[image]

ndesimone, 2015-11-08:
One other thing, the system won't let me take the latest update (the button is labeled "UP-TO-DATE").  I assume this is because 28 < 34 (although the dates are correct).

Dan-in-CA, 2015-11-08:
Thanks, I see the same thing here.
I'll check it out and get back to you.

Dan-in-CA, 2015-11-13:
OK. 
I just pushed an update to the SIP repository that fixes the problem. Let me know if you have any further issues with it.

Thanks for reporting it.

closed 2015-11-13 by Dan-in-CA

ndesimone, 2015-11-13:
Dan,
This still isn’t working right for me.  It looks like the gap is closer in version number, but the version I’m running is 34, and the repo is coming up with 30.  Here’s a screen shot:

Dan-in-CA, 2015-11-13:
I should have mentioned that the number the repo shows will still be off until you do an update. After the update things should be in sync.

The way the version numbering works is based on a calculation of the total number of commits to the repo minus a number that is based on the commit count that existed when the major version numbers are manually incremented. This applies to both your current version number and what the system_update plugin reports from the repo. GitHub itself does not provide a version number only a unique hash for each commit.

If you are still unable to do an update from the plugin me me know.

ndesimone, 2015-11-15:
Dan,
Thanks for the explanation.  I’m in the software industry and have the same problem with Git and version numbers.  Back in the old days, I used to use the subversion change number as the final part of my version string.  Now that we use Git, the version ends up being something that is manually maintained (ugh).

I’m still not able to update via the plug-in.  I think this is because repo version (using the calculation below) < the running software version.  Here’s a screen shot of what I’m seeing:

FYI, this isn’t blocking me, as I don’t need the Italian translations :)  I’m just trying help you get this all working as expected.  I’m guessing:

- I could get the software updated fine if I use Git directly
- Once there are 5 more commits in the repo (taking effective version from 30 to 35), the plug-in will start working again.

Hope this helps,
Nick

Dan-in-CA, 2015-11-15:
Yes this helps a lot. 

I tested with an old rev and it worked here. Hopefully not too many users will run into this problem. As you mentioned, running "sudo git pull" from the command line should avoid the problem.

A bit of further explanation:
Originally the plugin used the git command "git rev-list origin/master --count --first-parent" to get the number of commits. The first parent option skips some commits:
"...and this option allows you to ignore the individual commits brought in to your history by such a merge", - from:
http://schacon.github.io/git/git-rev-list.html

Running the command "git rev-list origin/master --count" (without --first-parent) returns the total number of commits. 

When I updated system_updater I removed the --first-parent option. There might be a better way to get a rev number from GitHub based on the information from the link above.