Dan-in-CA/SIP #140: Weather based rain delay does not conform to some municipalities restrictions

mihugo, 2015-11-10:
I woke up to my sprinklers watering after a pretty good rain storm the other night and discovered that the rain detector plugin resets itself after good weather is detected.  For example my city has the following rules:
>
  Any of the following acts or omissions, whether willful or negligent, shall constitute the waste of water:
          ...
           G.       Irrigation of landscaping during rainfall or 48 hours after a measurable rain event.

I've attached 3 patch files for this.

The first is just a display hack that shows the number of hours that the rain delay is active for on the main screen--much like  it shows the weather adjust.  I wasn't sure how best to convert seconds to hours/minutes so my method may not be the cleanest.
 
[templates-home.html.patch.txt](https://github.com/Dan-in-CA/SIP/files/31335/templates-home.html.patch.txt)

The second adds another option to the rain delay configuration: "Reset Delay on good weather".  It defaults to off (unchecked) which is the same functionality as current. It also should properly handle an older json config file so it users can update without issues.

[template-weather_adj.html.patch.txt](https://github.com/Dan-in-CA/SIP/files/31338/template-weather_adj.html.patch.txt)
[plugins-weather_adj.py.patch.txt](https://github.com/Dan-in-CA/SIP/files/31337/plugins-weather_adj.py.patch.txt)

Clearly this patch might mean that the plants will get under watered in the event of a  small showers. 
If there is a better way to submit the patch files let me know.  
I have not tried to deal with translations but if necessary we can work on that as  well.

Dan-in-CA, 2015-11-10:
Thank you!
The rain sensors I have seen don;t reset for a few days after a rain event. I usually set a delay manually from the home page when it rains.

We have the 48 hour restriction here also and your patches are a great addition to the plugin. 

I will integrate them into the plugin and update the plugins repository.

mihugo, 2015-11-10:
Glad to be of help. Great to see the work will be incorporated.   The default is actually checked and not unchecked as I wrote above. Either way the default is the same function as current with resetting on getting weather. One file  is for the main repo for the home page and not in the plugins.

Dan-in-CA, 2015-11-13:
I just pushed the merged files to the SIP and SIP Plugins repositories. I haven't done extensive testing but it looks like you did a nice job with the updates.

Thanks again.

closed 2015-11-13 by Dan-in-CA

mihugo, 2015-11-13:
THANKS!
Just updated my system from the repo and it looks fine.  I do see where I left out one tiny change adding a "h" to the end of the time delay print out on the main screen but not enough of a tweak to worry about now.

I also had trouble updating the plugin via the browser and ended up uninstalling and re installing the one plugin. Didn't take the time to debug yet.

Glad  to be of help here  with these changes.
Thanks

Dan-in-CA, 2015-11-13:
Oh, I took the 'h' off the end.
It didn't seem to make sense when the time was something like "4:56". I think everyone will get the meaning.