Dan-in-CA/SIP #142: SIP refactoring - RIMCO work

Pelado-Mat, 2016-01-07:
Hi, 
     Im looking to use OSPY/SIP with a custom HW based on relays controled with a I2C bus (I need something like 20 stations + pump ) and I have some ideas to add some pressure sensors, well water level and even more pumps to add fertilizer while irrigating.

 I started an attempt to refactor the code and encapsulate all the station control logic in a class to enable an easy way to support different HW configurations, but i found the work of Rimco, (https://github.com/Rimco/OSPy/tree/refactor) that it seems to be complete.

 Are there any plans to integrate Rimco's work to the project?

Dan-in-CA, 2016-01-07:
Some of Rimco's early work is already incorporated into SIP. Later he made some changes that were not compatible with features such as the ability to integrate with Samer Albahra's mobile app.

At this point there are no plans to integrate Rimco's fork into SIP.

Pelado-Mat, 2016-01-07:
Thanks Dan.
 
Are you interested in a "light" refactoring of the code to encapsulate all HW related stuff in a class or module? 

That way it would be easier to implement different HW configurations without  the ```gv.use_gpio_pins = False``` hack as relay_module uses.

Dan-in-CA, 2016-01-07:
Yes, I think that would be very useful.

nathantsmith, 2016-01-08:
I created custom boards with Lora radios that communicate  with some custom solar powered moisture/temperature sensors for a wine vineyard. I was planning on adding some additional functionality would either of you be interested in adding wireless capabilities, or a generic sensor input method for custom applications as such as this?

Pelado-Mat, 2016-01-08:
@nathantsmith, Im interested in adding sensors input, but I have already my hands full to cooperate on that.

Is anyone interested in checking an early version of the code? As I do not have the OSPI HW, only the relay board control Plugin is tested.

https://github.com/Pelado-Mat/SIP/tree/Control-Plugin

I would like some feedback as I'm interested in pushing everything upstream.

Regards

nathantsmith, 2016-01-08:
Do you need someone to roll out a hardware design?

Pelado-Mat, 2016-01-08:
@nathantsmith, thanks for the offer, but I do not need more than GPIO expansions at this point.

Dan-in-CA, 2016-01-08:
@Pelado-Mat,
I will test your code and let you know how it goes.
When you say that you don't have OSPi hardware, are you referring to the OpenSprinkler board?. I'm guessing you do have a Raspberry Pi.

@nathantsmith,
Can you provide a little more detail about your project?

Pelado-Mat, 2016-01-08:
@Dan-in-CA, I have a raspberry and the relays but I do not own the OpenSprinkler board.
 I realize that I could just try the ospi module using a couple of leds to check that the output is reasonable. I will try that during the week end.

Dan-in-CA, 2016-01-08:
@Pelado-Mat,
I have a Pi 2 and an 8 channel relay board and also a test rig with an OSPi board. I can test on both.
So far things seem to be working. 
The only problem I have found is that when the relay board is connected according to the docs for the original relay_board plugin (https://github.com/KanyonKris/relay_board/wiki) the relays are not activated in the same order:
S01 = relay 8
S02 = NA
S03 = Na
SO4 = relay 1
...
Looking at the defaultParameters['pin_map'] in relayBoard.py using RPi.GPIO I would guess pin 4 would = relay 1 but this does not seem to be the case. How are you connecting the relays?

Pelado-Mat, 2016-01-08:
Dan,
       I'm not following the same pins as KanyonKris as I'm reserving the
i2C pins and some others for future use. I think that in a general release,
the default configuration should be the same as the relay_plugin.

Anyway, the default pins that are actually used for the board are defined
in __autoconfigure_params

'relay_pins': [7, 8, 10, 11, 12, 13, 16, 15],  # Pins Conected with the relays.

This is saved in /data/RelayBoardControl.json.

Regards

bkoblenz, 2016-01-08:
Nathan,

I am interested to see what you built.  Can you email me directly at
brian.koblenz@bdkconsulting.com

Dan-in-CA, 2016-01-08:
Matias,
Some test results:
Using the pins you have configured causes relay 2 (pin 8) to chatter like crazy when the Pi is booted and relay 3 (pin 10) comes on momentarily also on boot.

Pelado-Mat, 2016-01-09:
Dan,
         you are right those pins are shared with the serial port, I should
avoid them.

Not taking into account the pin issue, are you OK with the direction I'm
taking with the refactoring?

As I see it, the next steps should be:
 - Debug and get ready the OSPi HW ControlPlugin
 - remove  sd['nbrds'] and sd['nst'] then replace with
len(gv.scontrol.stations)/8 and  len(gv.scontrol.stations)
   BTW, why the number of extension board is so important? Couldn't we just
use the number of stations?
 - Add/Modify "Options" webpage to:
       Change the ControlPlugin
       Change the Control Plugin Options
     * this should be difficult to me, as I have 0 experience with web.py
and webpages development... anyone interested in the challenge?
- Copy the Plugins logic to enable the downloading and updating of
ControlPlugins.
- Test and then test some more....
- Merge with master?

Further down the road:
  - Create an standard way to let plugins interact with the HW.
      The goal would be to enable plugins like "relay" to be created
without having to touch the main code base.

Dan-in-CA, 2016-01-09:
Matias,
I think it looks good so far.
Perhaps we should continue the discussion off line. Could you send me your email address? You can send it to:
dkimberling59@gmail.com

I will help with the webpage development. There are some other changes in the works for the web UI.
Plugins are an important part of this project. Making plugin creation easier is a goal.

Dan-in-CA, 2016-08-19:
Cleaning up repo after needing to spend time on other things.

closed 2016-08-19 by Dan-in-CA