Dan-in-CA/SIP #143: Feature request - flow meter add in

mattip, 2016-01-23:
Thanks for this wonderful application.

I have a Seed Studio flow meter http://www.seeedstudio.com/depot/G12-Water-Flow-Sensor-p-635.html hooked into my raspberry pi.

I am using wiringpi like the code posted here https://www.raspberrypi.org/forums/viewtopic.php?f=37&t=94656 to set up an interrupt handler and increment a counter. I see how I can write this as a so and read the counter from Python. I would like to get feedback in SIP about overall flow rate and expected vs actual flow rate (to detect leaks). What would be a strategy to get this into SIP? I see the way the clock time is updated, should I hook into that and update another div under it?

Note that I have not calibrated the flow meter, from some preliminary tests it seems the meter peaks out at about 100 pulses per second, so this might not be an accurate way to measure water quantity rather a qualitative go-no go check of the system, an arduino and maybe a better meter may be needed to get accurate measurements, but that is a different issue.

Dan-in-CA, 2016-01-23:
Welcome,
Thanks for your interest in the SIP project.

There is a user in Chile who is currently refactoring some of the SIP code to  make it easier to interface various hardware with the Raspi and SIP. He is also planning on using flow meters in his system.
It would be a good idea if you could post your feature request on the SIP discussion forum:
http://nosack.com/sipforum/index.php

I will contact Matias as I think he may be interested in working with you on this.
There may also be others who would have an interest in this.

bkoblenz, 2016-01-23:
If you use pigpio as the gpio interface, then you can set up a callback
function (called wind_cb) below and that callback function has a tally
method so you dont have to worry interrupt handling etc.

Just a couple of lines of code in your loop taking samples.  (Assumes there
was an initial call to set wind_cb prior to the loop.

                        flow = wind_cb.tally()
                        wind_cb = pi.callback(gv.pin_map[3],
pigpio.FALLING_EDGE)

Tally gives you the number of "clicks" since the last time wind_cb was
defined.

Pelado-Mat, 2016-01-25:
I plan to add a pressure sensor to do a go-no-go  check, as my irrigation water comes from a pump and a cistern I must be sure that the pump is actually moving water and not over-pressure the system.

 When we finish the work with Dan, I plan to modify the pressure sensor plugin to log and plot the pressure someway, also to stop all the stations if the pressure drops too much or get to high.

 A Flow meter plug in could work in a similar way, it should be easy modification of the pressure adj plugin using bkoblenz code.

shrakrak, 2016-04-23:
Has anyone eventually got around to writing a flow meter plugin. 
i am also interested in this feature and i wanted to check if something is available before i start my own solution.

BTW -  i think a flow meter should be part of the core application like the rain sensor. The reason is that it can effect many system functions like logging, active display of flow, accounting. it can also be used to detect solenoid malfunction and even detect leakage/clogging per station (comparing actual station flow Vs expected or historical flow data)

any opinions ?

DaveSprague, 2016-04-24:
For what it's worth, three years ago I grabbed a copy of what was called OSPi, at the time, and hacked in support for reading data from an arduino board that interfaces to three (or more) of the Seeed Studio flow sensors.  I was in a hurry at the time and it's worked so well and reliability that I haven't taken the time to merge this into the newer versions.

I agree that a flow meter touches several subsystems of the application so it's maybe be best to build support into the core app.

I'm happy to provide more details if that would be useful.  Here's an overview:

1) the Arduino code, which I'm happy to share if it's useful, keeps track of multiple pulse counters for each flow sensors.  Each time the RPi code reads a counter, via a USB serial interface, it resets that counter back to zero.

2) The RPi code reads the first counter for each flow sensor every few seconds to provide a real-time flow rate display. I use the second counter for each flow meter to track the flow during a program run (by reading that meter's counter it at the start of a program to reset it, the reading the total number of pulses after the program completes to determine the total flow).  I then use a third counter to track the amount of flow through each flow sensor over a twenty-four hour period to check if any of the irrigation system's valves is not closing properly.

3) I added the usage info the the log and also created a separate page to view the current flow rate of each channel.  At the time, I didn't have enough knowledge or time to integrate that into the primary display pages of the app.

I don't do any automatic checking of usage versus history or automatically checking a bad valve but I agree those would be great features to have.  I do email the recently log entries to myself once day to so that when we're away we can check that things seem to be working okay.

The reason I implemented multiple counters per flow meter is that I wasn't sure if the pulse count that was read every few 2-3 seconds for the flow rate display could be accumulated accurately enough over a longer period to get the total flow.  Looking at the usage per day though, there's enough unexplained variation in usage for a given program that it' s not clear that this is necessary.  I'll try to provide some more quantitive info on the amount of variation in usage from day to day for a particular program.

Another advantage of having multiple counters per flow meter is that it kept the RPi code somewhat simpler since I could devote a separate counter to each function (real-time flow, program usage, and valve malfunction / leak-detection).

I briefly considered implementing the pulse counting on the RPi using pigpio or similar library but at the time decided that using an Arduino to maintain the pulse counts was simpler for me -- particularly since I planned to maintain multiple pulse counters per flow meter.

Dave

Dan-in-CA, 2016-04-25:
Hi Dave,
This sounds very useful. I would be interested in more details.
You can post info on the SIP forum at:
http://nosack.com/sipforum/index.php
or email me at:
dkimberling59@gmail.com
or post code on GitHub and provide a link.

Thanks.
Dan

jakkuvaavinash, 2016-08-16:
the exact code for the pulse counting was not described in the fallowing so please get a valid code for water flow meter YF-S201 Hall Effect Water Flow Meter / Sensor so that will help to my project please refer me as soon as possible thanking you 

NOTE : PYTHON PROGRAMMING FOR RPI

avinashvera@gmail.com
8985797658
please contact me as soon as possible

Dan-in-CA, 2016-08-16:
I am not sure what code source you are referring to.
Can you please provide more detail?

Dan-in-CA, 2016-08-16:
Also, the YF-S201 flow meter requires 5V to work. 
The GPIO pins on the Raspberry Pi are 3.3V max. You will need to connect the flow meter to something like an Arduino.

DaveSprague, 2016-11-23:
Following along from my previous post in this thread from back in April, here's the Arduino code I use to count pulses from the flow meters.  It uses a library called from here: http://playground.arduino.cc/Main/PinChangeInt, although I used a version of this from 2013 so I'm not certain it will work with the current version without modifications.

This code is written for three flow sensors -- one for each valve in my system.  I maintain four counters for each flow meter so that I could accumulate counts over different time durations.  I did this because I wasn't sure if I read the sensors every few seconds and accumulated those values in the ospi python code would be accurate for looking at total flow over the course of a few hours.  It also makes the python code cleaner in cases where I wanted to keep logs of the total flow during a ospi program even or over a 24 hour period for each channel in addition to the flow during a particular scheduled program even.

I'm not sure this approach is necessary given the accuracy of the flow sensors but I haven't done any analysis of this yet.

I'll send the python code along some time soon, once I've had a chance to relearn it again and put a bit of documentation together.  It was based on a version of the ospi code from back in 2013.

`
// PinChangeIntExample, version 1.1 Sun Jan 15 06:24:19 CST 2012
//  See original example for more details on defines, etc
//#define NO_PORTB_PINCHANGES // to indicate that port b will not be used for pi
n change interrupts

#define NO_PORTC_PINCHANGES // to indicate that port c will not be used for pin 
change interrupts
#define NO_PORTD_PINCHANGES // to indicate that port d will not be used for pin 
change interrupts

// #define       DISABLE_PCINT_MULTI_SERVICE

#include <PinChangeInt.h>

int p1 = 8;  // digital pins 8, 9, and 10 are all on Port B on the Uno
int  p2 = 9;
int  p3 = 10;
int  led = 13;

// There are four counters maintained for each of three flow sensors.
// This allows for keeping track of the total flow over different timeframes
// e.g. 5 seconds, 1 minute, full program run, 24 hour period
long interrupt_count[4][3] = {{0L, 0L, 0L},
  {0L, 0L, 0L},
  {0L, 0L, 0L},
  {0L, 0L, 0L}
};
long unsigned startTime[4];
long elapsedTime;
boolean LED_State = HIGH;

int inByte = 0;

void quicfunc1() {
  digitalWrite(led, HIGH);
  interrupt_count[0][0]++;
  interrupt_count[1][0]++;
  interrupt_count[2][0]++;
  interrupt_count[3][0]++;
};

void quicfunc2() {
  digitalWrite(led, HIGH);
  interrupt_count[0][1]++;
  interrupt_count[1][1]++;
  interrupt_count[2][1]++;
  interrupt_count[3][1]++;
};

void quicfunc3() {
  digitalWrite(led, HIGH);
  interrupt_count[0][2]++;
  interrupt_count[1][2]++;
  interrupt_count[2][2]++;
  interrupt_count[3][2]++;
};

long flow_milliliters(long pulse_count, long milliseconds) {
  if ((pulse_count * 1000 / milliseconds) > 1000) {
    return -1;
  }
  long milliliters = pulse_count * 1000L * 2L / ( 11L * 60L);
  return milliliters;
}

void establishContact() {
  while (Serial.available() <= 0) {
    Serial.println("0,0,0,");   // send an initial string
    delay(300);
  }
}

void setup() {
  pinMode(led, OUTPUT);
  digitalWrite(led, LOW);

  pinMode(p1, INPUT);
  PCintPort::attachInterrupt(p1, &quicfunc1, FALLING);  // add more attachInterr
  upt code as required

  pinMode(p2, INPUT);
  PCintPort::attachInterrupt(p2, &quicfunc2, FALLING);

  pinMode(p3, INPUT);
  PCintPort::attachInterrupt(p3, &quicfunc3, FALLING);

  startTime[0] = millis();
  startTime[1] = millis();
  startTime[2] = millis();
  startTime[3] = millis();
  interrupts();
  Serial.begin(9600);
  //  while (!Serial) {
  //    ; // for Leonardo board
  //  }
  //  establishContact();
}

void loop() {
  digitalWrite(led, LOW);
  long count;
  if (Serial.available() > 0) {
    // get incoming string and convert to an integer for which counters are requested
    digitalWrite(led, HIGH);
    char c  = Serial.read();
    int freq = c - '0';
    int req = freq;
    int rst = 1;

    // bits 0-2 determine which of four counters is requested.
    // if bit 3 is low (default) we'll reset each of the four counters when
    //   reading them, otherwise (if bit 3 is high) we don't reset the counters
    if (freq >= 8) {
      rst = 0;
      req = freq - 8;
    }

    elapsedTime = millis() - startTime[req];
    String out_str = "";
    noInterrupts();
    // read the flow amount (in milliliters since the counter was last reset)
    //  for each of three flow sensors
    for (int i = 0; i < 3; i++) {
      count = interrupt_count[req][i];
      if (rst == 1) {
        interrupt_count[req][i] = 0;
      }
      String flow = String(flow_milliliters(count, elapsedTime), DEC);
      out_str += flow;
      out_str += ", ";
    }

    if (rst == 1) {
      startTime[req] = millis();
    }
    interrupts();

    // the output string are the values of the total flow for each flow
    //  sensors since this level of counter was reset, followed by the
    //  elapsed time since the flow sensors was reset.
    //  The serial input value as well as it's decoded values (the requested
    //  counter and the Reset flag) are also returned for debugging purposes.
    out_str += String(elapsedTime, DEC);
    out_str += ", ";
    out_str += String(freq) + ", " + String(req) + ", " + String(rst);
    Serial.println(out_str);
    digitalWrite(led, LOW);

  }
}
`

Pelado-Mat, 2016-11-24:
I still plan to add flow sensors to my installation, but Im busy with other stuff right now.

My to cents:
To get the arduino to connect to SIP, you could use a plugin as I wrote for the pump_control.
 
I connect the arduino to the pi via i2c and get the sensor readings of a pressure transducer (among other things).
See the plugin and arduino code from: https://github.com/Dan-in-CA/sip_plugins/tree/master/pump_control

 If you decide to go for that route:
 - The i2c code on the arduino side was kind of tricky to get right, specially when you have to send floats or ints via i2c, after reading tons of documents I came across https://github.com/MikeOchtman/Pi_Arduino_I2C that point me to the right direction
 - If you power the arduino from any other source.... connect the GND of the ardunio to de GND of the PI. (I fall into that....) 

BTW, every time I start thinking about this I  feel that SIP do not have the tools to really take advantage of sensors readings. I imagine some nice graph where I could see the different stations running overleaped with my sensors readings (pressure and/or water flow, etc).
 One path could be add some infrastructure to SIP to make that kind of graphs, using something like http://www.flotcharts.org/ (way over my current programming skills)
 Other way could be using @DanielCasner mtqq plugin to connect SIP to something like Domoticz, or PiDome and get the flow meters data  into that with [mySensors](https://www.mysensors.org/)

DaveSprague, 2017-04-07:
I'm doing a first cut adding Flow Sensor support, based on a simplified version of the implementation I described above and ported to the current code base.  The initial version will support flow sensors connected to an Arduino and accessed via USB serial.  It will show the real-time flow rate and flow amount while programs (including Run Once Programs) run and I plan to have it add total water usage to the log for each program run.

My plan is to release an early, rough version in the next several days.

I have included a stub that simulates flow sensors so that others can try it out prior to actually hooking up sensors (it just generates some pulse count values rather than accessing an Arduino).  I'm also trying to make it general enough that it would be fairly easy to connect flow sensors to directly to the RPi.

My assumption for this first version is that there's a flow sensor for each valve.  My own system only has three valves so that's how I've implemented it.  For larger systems it may make sense to have one shared flow sensor on the supply line that's shared by multiple valves or some other configuration like that.  There should be a fairly simple way to support this type of configuration with some settings options.

One thing I noticed is that the real-time update that's used to show elapsed time when running programs isn't used during Manual operation of the valves so we'd need to somehow add that into the Manual mode in order to be able to see flow rates/amounts when running in Manual mode (which I think would be quite useful).

I agree with one of the comments above that graphing would be useful.  My thought is that the webpage associated with the flow_sensor plugin would be a place to show these graphs.  It would access log data and ideally be able to show tables and graphs of water usage over different time periods (weekly, monthly, yearly?).

Another feature I'd like to implement is a way to look for stuck or leaking valves by watching for water flow during times when there shouldn't be any.

Let me know your thoughts on what features/configurations would be useful.

Dan-in-CA, 2017-04-09:
This sounds really cool!
One question I have is how to handle sensors that are some distance from the pi. I have been working with a USB to RS485 serial adapter at the Pi and an RS485 to RS232 adapter attached to an arduuino micro. The micro has a second hardware serial port which makes this easy. of course a lot of folks like to use wireless and since the latest Pi 3s have built-in wifi it should not be too hard.

Regarding data graphing, I have been testing chart.js. and have been able to make a plugin page with nice graphs. There is an intro to chart.js with some examples here:
https://www.sitepoint.com/introduction-chart-js-2-0-six-examples/

Looking forward to seeing what you have put together.

DAn

DaveSprague, 2017-04-11:
That charting package looks great.

Good question about remote sensors.  I really haven't given that any thought so far.   In my installation, I have my flow sensors installed right after each valve and since the PI (with the OSPi board, in my case) need to be fairly close to the valves then I assumed the flow sensors would also be close  -- although the valves solenoids can probably be driven over a longer distance than the return signals form the flow sensors can travel.  So yes, I would think we could/should accommodate both RS485 and WiFi connected Arduinos.

Do you power the Arduino over the RS485 connection or just provide power to it locally?

Just finished writing a simple version of the Arduino code that maintains pulse counters for up to 8 flow sensors.  It could be expanded beyond that number fairly simply.  I'm using the EnableInterrupts library (https://github.com/GreyGnome/EnableInterrupt) which which can use up to 18 inputs on an Arduino Uno.

If you're trying to accurately monitor the flow from several valves/lines at once, this code may not be fast enough so if that's an important issue, we may need to use the HiSpeed mode of the EnableInterrupt library or write some other custom Arduino code.

My current implementation has the Arduino code just count pulses for each flow sensor and allows all the counters to be reset at once.  The conversion to flow rate in Gallons or Liters per second and total flow amounts in Gallons or Liters is done in the flow_sensor plugin.  I partitioned it this way since I thought we might want to support flow sensors connected directly to the Pi, using an interrupt routine to count the pulses.  In some ways it would be cleaner to have those conversion details encapsulated in the Arduino code but at this point, that's not how I've structured the code.

As you mentioned above, the flow sensors are designed to run at 5v so maybe the directly interface to the Pi isn't something we really need to support?

My testing is using the 1/2" and 3/4" plastic flow sensors similar to this one:
http://wiki.seeedstudio.com/wiki/G1/2_Water_Flow_sensor

Dave

Dan-in-CA, 2017-04-11:
At this point I have been connecting to the arduinos over standard 18guage sprinkler wire. I use 2 wires for the serial communication and 2 wires carrying 12V DC to power the arduinos. I have been using Arduino micros. They are small and have 2 hardware serial ports. They are soldered to an Adafruit 1/2 size perma-proto board along with the RS485 to RS232 adapter and the whole thing fits into a 1 gang weather proof junction box which makes a nice inexpensive water tight enclosure.
[image]
Of course there are larger weather tight junction boxes that could b used for the Arduino Uno.

I have one of the 1/2 inch plastic flow sensors that I can use for testing.

I will look into adding the real-time update for the manual mode.

DaveSprague, 2017-04-12:
Hi Dan,

That setup looks great.  I'll have to pick up an ardiuno micro to try out.
Great idea using the junction boxes.

I'll send you the current version tomorrow.  What's the best way to give
you the code -- I can do a push request but I can also just send you a zip
file of the full SIP directory if that's easier.  The changes from your
current version on github are small changes in the following files:
helpers.py, templates/home.html, templates/log.html, and webpages.py and
then of course the addition of the plugins/flow_sensors.py file.  Also
included will be the Arduino .ino file.

You'll probably want to change the style parameters of the Stations List
table so that the flow and total water usage values fit better.  I couldn't
locate where the width of the columns of that table are set.  Right now,
the flow rate and amounts, with units, are just tacked on to the time that
counts down each second and so they wrap around to a second line,
increasing the height of that table row while the program is running.

I'm only updating the instantaneous flow rate and usage values every three
seconds in hopes of getting somewhat more accurate flow rate numbers but
that's a parameter you can set within the plugin file.

I have not yet add any Options in to support the flow sensors.  For now,
those are just set up near the top of the flow_sensor plugin file (e.g. the
type of flow sensor being used, whether to give rates and amounts in Liters
or Gallons, etc).  I'm also at this point just supporting a direct mapping
from the eight flow sensor counters running on the Arduino and the first
eight valves.

Dave

Dan-in-CA, 2017-04-12:
If you let me know when the SIP code is ready I can clone your fork from GitHub. That way I can pull updates as things progress and you won't need to do anything special.
You can either send a zip of the arduino code or post it to a gist here in GitHub. Just click Gist at the very top of you repo page.

DaveSprague, 2017-04-12:
My branch of SIP with the flow_sensor support is here:  https://github.com/DaveSprague/SIP

The Arduino code is in a public Gist here:  https://gist.github.com/DaveSprague/88c2380297e0ae53605ac360299f1b6b

Near the top of the flow_sensors.py plugin you can set type type of sensor you're using (only 1/2" and 3/4" plastic Seeed types are supported now) and whether you want the units in Liters or Gallons.

It currently defaults to flow sensors connect to an Arudino that's connected to the Pi via the USB serial interface.  You may need to change the /dev/tty??? name in the flow_sensors.py plugin code to connect to a different serial interface.

Also, near the top of the plugin you can change from using Arduino connected flow sensors to using a very basic simulated flow sensors so that the program can be tested even if hardware sensors aren't connected yet.

When you run a Program or Run-Once program you'll see the flow rate and total flow amounts for each station updated every 3 seconds on the Home page.  It also adds the water usage amounts for each Program or Run-Once program to the log file.

Doesn't currently show any flow information for Manual Mode yet.

I've left in a bunch of print statements I used for debugging for now so you can use them to identify any issues.

Let me know how it goes and by all means give me feedback on and code issues or functionality issues.

This is my first open source code contribution!

Dave

Dan-in-CA, 2017-04-12:
OK. I have cloned your fork, copied the Arduino code and downloaded EnableInterrupt to my arduino library. The Arduino code compiles and I loaded it onto an Arduino Micro that I have on a breadboard.

I needed to install pyserial on the Raspi and the plugin is loading properly. I connected the arduino to the Pi with a USB cable and am getting:
```
Writing to Arduino
serial input from Arduino is: 0,0,0,0,0,0,0,0
```
So it looks like they are communicating but I have not connected the flow sensor yet.
I have not taken a close look at the Python code yet but this looks like a great start.

DaveSprague, 2017-04-12:
Yes looks good so far. Those eight zeros are the counter values for up to
eight flow sensors. You can see in the arduino code the pins I currently
have set for the flow sensors. As I recall, it's pins 5-12

Dave

Dan-in-CA, 2017-04-12:
I hooked up the flow sensor I have to the Arduino micro but I didn't get any reading (blowing air through the sensor). After switching to an Arduino Uno it is working.

There is probably some difference between the 2 Arduino models.

Dan-in-CA, 2017-04-12:
Looks like the Micro has a different set of interrupt-able pins.
I am getting a reading on the Micro with the sensor connected to pin 8 using your original code. It may be necessary to have different versions of the Arduino code for different arduinos.

DaveSprague, 2017-04-12:
You might want to try some different pins on the Arduino Micro.  The Arduino versions are a bit confusing to me but this page on the EnableInterrupts library talks about what pins can be used on different Arduino versions:

https://github.com/GreyGnome/EnableInterrupt/wiki/Usage#pin--port-bestiary

DaveSprague, 2017-04-12:
Yes, or if the Arduinos are "self-aware" enough we might be able to configure, at startup,  certain pins to be used based on what type of Arduino it is.

Dan-in-CA, 2017-04-12:
Not sure if it is possible to to determine the Arduino model automatically but there are ways to handle it in a single code base. That does not seem to an urgent matter.

DaveSprague, 2017-04-12:
Yes, agreed.  We can find a way to support a few popular Arduino models pretty easily I think.

DaveSprague, 2017-04-12:
The product description of the Arduino Micro, here: http://www.newark.com/arduino/a000053/atmega32u4-arduino-micro-dev-board/dp/63W3544
and the spec sheet you can download there seems to
say it's similar to a Leonardo so you might want to check it against this list of pins from the EnableInterrupts usage:

Arduino Leonardo
 Interrupt Type | Pins
 -------------- | --------------
 External       | 0-3 and 7
 Pin Change     | 8-11 and SCK, MOSI, MISO, SS (SS on 3rd party boards)

DaveSprague, 2017-04-14:
Were you able to get the Micro working okay?  I have a Leonardo which appears to be identical except for the form factor so I can do some testing here as well.  Here's a link I found on the Leonardo/Micro -- we may have to modify the serial code somewhat.

https://www.arduino.cc/en/Guide/ArduinoLeonardoMicro

Dan-in-CA, 2017-04-14:
Yes. The micro works using ports 8-11 with the code as is which is fine for testing. 

Considering that a user will need to open the Arduino file so they can upload it it would be easy to have (commented out) lines defining the pins for different models and instructions at the top directing the user to un-comment the line for their particular model.

DaveSprague, 2017-04-15:
I just noticed that in home.html, in the updateStatus function, around lines 69-70 I should have put in a test to make sure station.flowrate and station.flowAmt are available before I append them into the display string (to properly handle the case where the flow-sensor plugin isn't being used).  Do you want me to make that change?

shrakrak, 2017-04-15:
Hi Dave

i have some comments and suggestions regarding the flow meter support

1. Assuming you only operate a single line at a time you can use a single flow meter before the master valve to measure all lines. it is cheap and simple and doesn't impose a strict limitation. So it think this topology should probably be supported.

2. valve malfunction detection - i believe you mentioned this one already. it would be really cool to have that

3. line damage detection - This one is a bit more complex. using historical measurements we can estimate an estimated flow per line using averaging (again assuming single line at a time). If the flow exceed the average by X percent (configurable) some sort of notification is sent. This will allow to detect tears in the hoses or detached droppers.
Off course when you add outlets it will false detect a damage but that can be easily handled using some "reset" button. 

Thanks for picking up this feature.
it will be a great addition. 
Let me know if you need a beta tester for direct attached flow meter

Dan-in-CA, 2017-04-15:
Yes. Go ahead and make the change in your fork.
I am also thinking about possibly having some way to load plugin specific JavaScript code.

Right now the plugin files are kept in a separate Github repository:
https://github.com/Dan-in-CA/sip_plugins

The README  for that repo is used by the plugin manager to list the available plugins. When a user selects a plugin for installation the plugin manager downloads the contents of the folder for the selected plugin(s). There is a manifest file included with the other plugin files. The manifest lists the files that make up the plugin along with their locations in the directly structure. The plugin manager reads the manifest file and places the files in their proper locations. A similar process is used to remove the files if the user uninstalls  a plugin.

Until now, none of the existing plugins have had to modify the home page or other pre-defined pages but that should definitely be an option.

Another part of the plugin system is the file __init__.py in the plugins directory. It does the actual plugin loading when a plugin's group mode is set to executable.  That is how plugins can be enabled or disabled.

If it is necessary to modify the base code of one or more of the UI pages to make a plugin usable that is fine with me but it would be nice to be able to load plugin specific JS code that can modify an existing page. That would make code maintenance easier.

DaveSprague, 2017-04-16:
@shrakrak agree with your three points.  The single flow meter will probably be a popular option, even in cases where there's not a master valve but just a single supply line for some number of valves so we should have a fairly flexible way of assigning meters to valves.  I like your idea about line detection too.

@Dan-in-CA Yes, I'll make that change and let you know when it's available.  I agree with the desirability of allowing plugins to have javascript code or at least some fairly general way to allow plugins to display real-time data on the home page and to add columns to the log is very desirable.

One way to do this, at least for the log, would be to have a gv.log_columns dictionary that plugins can add entries to.  For example, the flow_sensor plugin would add a reference to it's program water usage variable (currently named as gv.plugin_data["fs"]["program_amounts"]) so the key for this entry "Usage" would be used as the column name and the value would be a reference to that "program_amounts" variable.  One complicating factor is that currently the program_amounts value is list with an entry for each flow_meter so you would need a way to designate that you want to select the specific amount value corresponding to the currently active valve/program.

Thanks for the description of the plugin architecture.  That's very helpful.  I'll work on getting the flow_sensors plugin compatible with it.

Dan-in-CA, 2017-04-16:
Dave,
I have an idea for a way that plugins can modify the pre-defined pages using JavaScript/jQuery to add new display features. It's a little complicated to describe and I have not tested it yet. I'll let you know more after I have a chance to do some tinkering.

The log data is already stored as a series of dictionaries in the SIP/data/log.json file. A plugin could add elements but there would still need to be a way to modify the log page to show the new data and possibly to add the use data to the timeline on the home page.

The log file is updated by the log_run() function in helpers.py line ~321. Perhaps we could add a signal to that function that would be broadcast when a line is added to the log. Then the plugin could update the log with it's own data. The log is updated when a station run is completed. The plugin could get the station ID from the log data, add the usage data and reset the usage counter. -- just some brainstorming. 

This is all very exciting. Sometimes a user will request a new feature but when someone actually gets involved to create the feature as you are doing I find it quite motivating. Also the exchange of ideas from different view points leads to innovations that I would not have even thought of.

Dan

DaveSprague, 2017-04-19:
Hi Dan,
I'm working on one way to allow plugins to add items to the log.  I'll send you that once I have something working and see what you think.

I'm not tied to this specific approach but it seems like it will do what we want it to do.  The basic idea is to have a variable in gv that I'm calling gv.logged_values that is a list of pairs.  The first element of each pair is the name of the value and the second element is a function that is used to compute the logged value at the time the log is updated to the log.json file.  (I'm using a list rather than a dictionary since the order is controllable which is important when you display the log.)

I'm initializing the list to be the default logged values as they are in the current code base but it will allow plugins to add other values to the log by adding other name/function pairs.  I'll modify the flow_sensor plugin to add it's water usage values.

I considered the signal approach you mentioned but we would still have the issue of the log.html file having to figure out, when displaying the log, what the name of the new values added to the plugins are so I think this approach is comparable to that one.  It also avoids having two writes to the log each time the log is updated.

I've got this new version working to the point of updating the logs properly.  Now I'm going to update the log display code to use the same approach rather than just having a fixed set of column names.

Once I have something that you can take a look at, I'll move on to making the flow_sensor plugin compatible with the plugin architecture and then hopefully start looking at the charting.

Dave

Dan-in-CA, 2017-04-19:
Sounds good! Can't wait to see what you've come up with.
I will be off-line for a couple of days so no rush.

DaveSprague, 2017-04-23:
I've made some good progress:

**Arduino Code Fix**
First, I found a bug in my Arduino code.  I was only using 16b counters for the flow meter pulse counters but 32b counters are required to avoid wrap-around when the water quantity goes above about 140 gallons.  I looked back at my previous implemtation and, sure enough, I was using 32bit counters for it.  I've updated the Gist archive of the Arduino code to use 32b unsigned counters.
https://gist.github.com/DaveSprague/88c2380297e0ae53605ac360299f1b6b

**Logged Values from Plugins**
I've completed the new implementation of the logging function that supports plugins adding values to be logged.  There's a new gv variable called gv.logged_values.  It is a list of values to be logged.  Each value entry is a two-entry list consisting of the name of the value (which is used for the key name in the json file and for the Column name) and a function that is used when the log is created to compute the value to be logged.  Here's an example of code in the flow_sensor plugin that places an entry in the log list to include the amount of water used by each program run:

`gv.logged_values.append( [_('usage'), lambda : '{:.3f}'.format(gv.plugin_data["fs"]["program_amounts"][gv.lrun[0]]) ])`

What this does is it looks uses the current station number, stored in gv.lrun[0], to index into the flow_sensor plugins gv variable that stores the water quantity amounts to get the one for the station that just finished running.  It takes that float value and converts it to a string with three digits to the right of the decimal point.

I also added code into the gv.py module that adds the proper values into gv.logged_values to replicate the default log format.

One restriction is that the anonymous function that computes the value to be logged has to return a string so if you need to log a numeric value or boolean you should convert it to a string within the anon. function.

This code is in a separate fork of the SIP project here:
https://github.com/DaveSprague/SIPdev.git

**Convert Flow Sensor plugin to be compatible with the Plugin Architecture**
Making good progress on this.  I have added the manifest and also added the code to implement a webpage for the plugin that allows you to set the type of interface to the flow sensors (only Simulated and Arduino Serial are supported now), the type of flow sensor used (only 1/2" and 3/4" plastic Seeed type flow meters are supported currently,  and whether to log/display Liters or Gallons.  Still tracking down a few bugs in this and have yet to test that the plugin can be loaded from the sip_plugins repository.

Dan-in-CA, 2017-04-25:
OK. I have gotten your updates and will start exploring and testing.

DaveSprague, 2017-04-25:
Thanks Dan.  My fork of SIP that's named SIPdev is the one that has the new stuff in it.  It looks for plugins in my fork of sip_plugins which is where the flow_sensors plugin is now.

Dan-in-CA, 2017-04-25:
Started doing some testing:
The plugin install went without a hitch. You have really made progress.

On the plugin settings page I switched from "simulated" to "Arduino-Serial" but it still seems to be running in simulation mode.

I am wondering about support for other flow sensors that users might try to use. Looking at Amazon there are several brands. One vendor has sizes from 1/4 inch to 2 inch:
https://www.amazon.com/DIGITEN/b/ref=bl_dp_s_web_14129847011?ie=UTF8&node=14129847011&field-lbr_brands_browse-bin

It might be good to allow the user to (optionally) set parameters for their sensor if it differs from the common types.

Haven't had a chance to dig into the code yet but things are looking good so far.

DaveSprague, 2017-04-25:
I'll do a check to see if I'm seeing the same issue on the switch from simulated to Ard.-serial and figure out what's causing it.

That's a great idea to support parameters as an alternate way to set up a specific flow meter.  I'll put that on the list of features to add soon.  It should be fairly easy.

DaveSprague, 2017-04-25:
Another feature I'm planning to add soon is to be able to associate flow sensors with valves in a fairly flexible way, e.g. for each valve you can specify the number of the flow sensor that should run when it's activated. The current setup would be [0, 1, 2, 3, 4, 5, 6, 7] and a system with a single flow_sensor on the master line to all the valves would be [0, 0, 0, 0, 0, 0, 0, 0].  That will probably give us enough flexibility for now?

Dan-in-CA, 2017-04-25:
Regarding parameters for different sensors; It might be good to include a "calibration" value.
For example a user could run a valve long enough to fill an e.g. 5gal bucket and use the sensor reading from that to set parameters for the sensor. That should make it fairly easy to set up a sensor without knowing it's specs.

DaveSprague, 2017-04-25:
Yes, I agree.  I've never actually tried testing on of the flow sensors yet to make sure the gallons I'm reporting are fairly accurate so it would be interesting to have someone try that at some point too.

I just tried the Simulated to Arduino-Serial and back to Simulated again and both cases seem to work as expected for me when I kept the flow_sensor type as the Seeed 1/2".  I did find one bug in the flow_sensor plugin that occurs when you select the 3/4" flow sensor.  In that case, you get an invalid key error when it starts reading the sensors because I had the name for it wrong in the flow_sensor.html file (I spelled it as Seed rather than Seeed three times on line 41.

Are you selecting a 3/4" flow sensor?  If so, that may fix the problem.  I just pushed a corrected flow_sensor.html up to my fork of sip_plugins so if you reload it that will fix that problem.

If you are using a 1/2" flow sensor, then we'll need to do some more debugging.  I still have some print statements in place from testing the savings of the settings.  When you save them, you should see something like this in the standard output:

```

1. settings are:  {'interface': 'Simulated', 'units': 'Gallons', 'sensor_type': 'Seeed 1/2 inch', 'rate_units': 'GpH'}
2. GET method in settings class
3. No flow_sensors.json file
4. my settings here are: {'interface': 'Simulated', 'units': 'Gallons', 'sensor_type': 'Seeed 1/2 inch', 'rate_units': 'GpH'}
5. 192.168.1.118:55701 - - [25/Apr/2017 11:25:53] "HTTP/1.1 GET /flow_sensors-sp" - 200 OK
6. 192.168.1.118:55701 - - [25/Apr/2017 11:25:53] "HTTP/1.1 GET /api/log" - 200 OK
7. qdict = <Storage {'interface': u'Arduino-Serial', 'units': u'Gallons', 'sensor_type': u'Seeed 1/2 inch'}>
8. settings : {'interface': 'Simulated', 'units': 'Gallons', 'sensor_type': 'Seeed 1/2 inch', 'rate_units': 'GpH'}
9. resetting flow sensors
10. values from Arduino on establishing serial port
11. 0,0,0,0,0,0,0,0
12. 
13. after update from qdict, settings = {'interface': u'Arduino-Serial', 'units': u'Gallons', 'sensor_type': u'Seeed 1/2 inch', 'rate_units': 'GpH'}

```
That last line -- line 13 above, that starts with "after update from qdict, settings =" should show the settings properly being set.  In this case, it's showing the settings being Simulated in earlier print statements (lines 1, 4, and 8).  In line 7 we see qdict returned from the setting Save action with the new settings and then the settings variable being updated to Arduino-Serial in that last line, line 13.

DaveSprague, 2017-04-25:
I also update the data/flow_sensors.json file with the new setting when they're saved.

Dan-in-CA, 2017-04-25:
I de-installed and reinstalled the plugin to get the updates.
It seems to be working as expected now. The sensor had been connected to a prototyping Arduino Micro on a breadboard with a bunch of other stuff connected to it. That may have contributed to the problem.
Now I have the sensor connected to an Arduino Uno without anything else connected.

Dan-in-CA, 2017-04-25:
Hmm...
After rebooting the Pi it is back to running in simulation mode with the  plugin set to Arduino-serial.
The flow_sensors.json file has Arduino-Serial as the interface so the setting is saved properly.

DaveSprague, 2017-04-25:
Hmm...let me try that and see if I can replicate the problem

DaveSprague, 2017-04-25:
Actually thinking about for a second, I believe the problem is that I only access the json settings file for the addon when I go to the settings page.  I need to do that when the plugin is initialized, otherwise, it's switching to the default values with is Simulated.  Should be a fairly quick fix

Dan-in-CA, 2017-04-25:
I think that's it. I went to the settings page and resaved then it started working.

DaveSprague, 2017-04-26:
Yes, I think so too.  I fixed the startup code in flow_sensors.py and also noticed a case where the quantity and rates units (e.g. gallons and gallons per hour) could get out of sync (e.g. Liters and Gallons per hour) so added a bit of code that I believe fixes that too.

I've uploaded the changes to sip_plugins and tested it some and it seems to be working properly now.  The very first time I reloaded the plugin and tried to run a Run Once program, I got this error:
```
Some Stations has been scheduled: [[0, 0, 0, 0], [1493139588, 1493139600, 12, 98], [0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0]]
192.168.1.118:55287 - - [25/Apr/2017 16:59:49] "HTTP/1.1 GET /cr" - 303 See Other
192.168.1.118:55287 - - [25/Apr/2017 16:59:49] "HTTP/1.1 GET /" - 200 OK
192.168.1.118:55287 - - [25/Apr/2017 16:59:49] "HTTP/1.1 GET /api/log" - 200 OK
Traceback (most recent call last):
  File "/home/pi/SIPdev/web/application.py", line 237, in process
    return p(lambda: process(processors))
  File "/home/pi/SIPdev/web/session.py", line 85, in _processor
    self._load()
  File "/home/pi/SIPdev/web/session.py", line 106, in _load
    d = self.store[self.session_id]
  File "/home/pi/SIPdev/web/session.py", line 250, in __getitem__
    return self.decode(pickled)
  File "/home/pi/SIPdev/web/session.py", line 210, in decode
    return pickle.loads(pickled)
EOFError
```
But I haven't been able to replicate it so keep an eye out for it.

DaveSprague, 2017-04-26:
Just noticed that this "pickle EOFError" error is similar or the same as the one noted here:  https://github.com/Dan-in-CA/SIP/issues/37

Dan-in-CA, 2017-04-26:
I see that error on rare occasions. I have never been able to find the cause or a solution.

DaveSprague, 2017-04-28:
Some thoughts on what I'd work on next.  Let me know which would be best to prioritize.

1) Take a pass at cleaning up / refactoring the flow_sensors.py code so it's more readable.
2) Start adding support for data gathering that will eventually be used for charting.
3) Adding support for custom flow_sensor parameters to permit using a wider range of sensors and supporting calibration.
4) Adding some documentation linked to the Help button that would describe how to set up flow sensors and use the plugin.

Some initial thoughts on the data gathering (#2 above).  I think we should add a Signal for End of Program (which I don't think exists currently?) and also for End of Day.  This would allow us to capture the water usage on each station and accumulate it to get Daily, Weekly, and Monthly totals.  We'd then maintain a separate log via a json file that stores the current day's usage, and the last 7-14 days, last 10-15 weeks, and last 12-24 months of water usage.  As part of this we can also track any water usage that occurs when a program is not running on a station to check for a valve that's stuck on.

Dan-in-CA, 2017-04-29:
Hi Dave,

As far as priorities, I think #3 would be good to work on. The calibration and  parameters for different sensors are closely related. Doing a calibration should provide the values needed (volume / ticks) to use the sensor without even needing to know the published specs.

Of course documentation in important but if there might be changes to how things work it might be a good idea to start an outline and develop toe details when things are more complete.

On #2, Agreed that signals that mark time intervals like  days, weeks, etc should be added to the core code. In fact another user is working on a plugin that will send notifications to e.g. a smart phone when selected events occur. See the SIP forum:
http://nosack.com/sipforum/showthread.php?tid=72&pid=275#pid275

He has used a signal that marks the end of a station's run. It might be good to include some signals in the flow_sensors plugin that could be used to notify a user of unusual flow rates. There is also the telegramBot Plugin that could notify a user of abnormal operation.

#1 cleaning up the code should make further development easier. I guess how you prioritize things depends on your  particular way of doing things.

I have been trying to get communication working over RS485 so I can connect to sensors that are not physically close to my Raspi. I have 7 valves that are in 4 locations around the house. The closest ones to the SIP controller are about 30 feet away (beyond the 16' max for USB). I think this situation is probably fairly typical. 
So far I am able to write from an Arduino micro to the Pi but can't seem to get it working in the other direction. Probably just not doing something right.  This should not involve much change to the plugin code. The RS485 adapter plugs into a USB port on the Pi and converts RS232 to RS485. On the Arduino side there is an adapter the translates from RS485 to RS232 so it is just basic serial communication. There would also need to be some sort of addressing method if there are 2 or more arduinos in the system.

DaveSprague, 2017-04-29:
Yes that sounds like a good priority for me to address items.  I'll start work on items #3 and #4 for now.

As for the end of day, week, month type signals one thing we'll have to think about is what order those should be generated in the case where the end of day and end of week and possibly end of month occur at the same time.  My first thought is that the end of day should be signaled first and then the end of week happens maybe a few seconds later, and then when necessary, the end of month would be send a few seconds after that??

The other option would be to just send an end of day signal and let the code/plugin that uses the signal go through and check if it's also the end of week and end of month if it needs to.  That way it can handle those situations however it chooses.  An example would be where at the end of each day I compute the daily water usage and then for the end of week I use that day's total plus the previous six day totals to compute the weekly water usage.

On the RS485 serial interface, you should check that you're flushing the output buffer on the Pi side after you write something?  Here's the code I'm using (edited slightly for clarity) in the flow_sensors plugin to write to the arduino and then receive data back.  I'm not completely happy with having those time.sleep calls in there but they are working for now.

```
ser = serial.Serial('/dev/ttyACM0', 9600, timeout=1)
time.sleep(0.1)
ser.write("RS\n")
ser.flush()
time.sleep(0.1)
line = ser.readline()
```

mattip, 2017-06-03:
Hi. I would like to try this out, it would be great just to get feedback that water is flowing in the system when a program is running, no need for calibration, just a go/nogo. 

What should I do to try out the code?

DaveSprague, 2017-06-04:
Hi Matti,

If you'd like, you can use my modified clone of SIP here: https://github.com/DaveSprague/SIPdev

It has some changes to the base SIP code that allows the real-time gallons or liters per hour to display, on the home page, while a valve is running and it modifies the log code to permit plugins to add columns to the log.  It also points to my own version of the plugins repository so that you can add the flow sensors plugin that handles the interface to the flow sensors.

The only configuration that's currently supported is one or more flow sensors that interface to an Arudino and then the Raspberry Pi that's running SIP reads the flow data from the Arduino via a USB serial interface.  The Arduino code is located here: https://gist.github.com/DaveSprague/88c2380297e0ae53605ac360299f1b6b

The Arduino I use is an Uno so if you use a different type, there might be some debugging to get the connection working and you might need to attach the signals from the flow sensors to different input pins but I can work with you on that.

Since you only want to see whether water is running or not, you can just keep the flow sensor plugin default for the type of valve.

Do you expect to use one flow sensor per valve or a single flow sensors on the main supply?  I haven't yet added the code to the flow sensor plugin options to support the later yet but I can do that in the next couple of days if that's how you expect to use it.

Thanks,

Dave Sprague

mattip, 2017-06-04:
body p { margin-bottom: 0cm; margin-top: 0pt; } 
  
  
    Cool, I will try it out. I only have one flow meter, so no extra
    code needed.
    Thanks for putting in the effort
    Matti
    
    On 04/06/17 19:47, David L Sprague
      wrote:
    
    
      Hi Matti,
      If you'd like, you can use my modified clone of SIP here: https://github.com/DaveSprague/SIPdev
      It has some changes to the base SIP code that allows the
        real-time gallons or liters per hour to display, on the home
        page, while a valve is running and it modifies the log code to
        permit plugins to add columns to the log. It also points to my
        own version of the plugins repository so that you can add the
        flow sensors plugin that handles the interface to the flow
        sensors.
      The only configuration that's currently supported is one or
        more flow sensors that interface to an Arudino and then the
        Raspberry Pi that's running SIP reads the flow data from the
        Arduino via a USB serial interface. The Arduino code is located
        here: https://gist.github.com/DaveSprague/88c2380297e0ae53605ac360299f1b6b
      The Arduino I use is an Uno so if you use a different type,
        there might be some debugging to get the connection working and
        you might need to attach the signals from the flow sensors to
        different input pins but I can work with you on that.
      Since you only want to see whether water is running or not, you
        can just keep the flow sensor plugin default for the type of
        valve.
      Do you expect to use one flow sensor per valve or a single flow
        sensors on the main supply? I haven't yet added the code to the
        flow sensor plugin options to support the later yet but I can do
        that in the next couple of days if that's how you expect to use
        it.
      Thanks,
      Dave Sprague

mattip, 2017-08-13:
@DaveSprague I have finally gotten around to testing this. Thanks for all the work, it certainly made it easy to get started. Here are some thoughts:
- I did use the GPIO module + multiprocess.Process to set up a callback in another python process to increment a counter
- I only have one flowmeter and am returning [value]*8 from read_flow_counters. Won't that give strange values for the watering amount on each of the 4 stations?

I will let it run for a few days to see if the reporting is useful. Already it has achieved my goal which was to verify that a running program actually opened the valve, my whole setup is a bit unreliable.

[12 more replies]

closed 2019-12-30 by Dan-in-CA