Dan-in-CA/SIP #145: Which version/release, GPIO options

Tony-in-Oz, 2016-03-21:
Hi
I am looking to deploy a Sprinkler and LED lighting system on my existing Raspberry that manages my Solar system.  I have a Model B  with 26 pin header with no GPIO pins previously in use.  And I'd like to use the same system for other automations/recording as well.  So likely I'll need more GPIO i/o.  Eg measuring water flows, Water tank levels, power consumption, natural gas use and more.

At the moment I can easily get 12 relays working using the relay_board plugin and have 4 working now in a test setup.  But leaves almost no i/o for other uses.

Questions
Is there a plugin for use when using the MCP23017 chips on the Ic2 bus?
It seems these can easily expand at 16 port a time.

Also there are several 'Opensprinkler' versions on GIT.  What is the difference between each -
- SIP
- OSPi
- OSpy

Obviously the "OpenSprinkler" firmware version uses dedicated hardware to achieve it's i/o ports.  Although there are at least 2 different board suppliers, Rays and TIAO.  Personally I like the idea of relay contacts protected by a circuit breaker should a short or other event occur.  I want to keep my 24v ac separate form 3/5v logic.

Interested in some feedback and discussion

kind regards

Tony

Dan-in-CA, 2016-03-21:
Hi Tony,

There is a guy in Chile who is developing code to use i2c and other controller interfaces with SIP. You might find his post on the SIP forum interesting:
http://nosack.com/sipforum/showthread.php?tid=25

He has things working now but we are still in the process of developing the user interface.

SIP started out as a Python port of the OpenSprinkler Arduino code. It was originally called  OSPi. Eventually users started asking for relay board support and  the OpenSprinkler project went commercial. At that point I decided to rename the Python project and make it more relay board friendly. The goal of this project has always been to be easy for DIYers to modify.

The OpenSprinkler Arduino C/C++ code has been modified to run on the Raspi as well as the Arduino but is not as easy to tweak as SIP's Python code.

There is also the option of using 74c595 shift register chips with relay boards.This works without modification of the program code but requires the relay boards to be active high.

Dan

bkoblenz, 2016-03-21:
I have built a system that was based on sip, but I have extended it and
modified it significantly.  In particular, I have eliminated some of the
plugin flexibility and architecture because I could not effectively test
what I was doing with all the plugins.  That is the bad news.

The good news is that I have tested integrated hw and sw and am currently
providing early versions of the hw and sw to some "friendly" users.  I have
not yet decided how broadly to make things available.

The key features of my extensions include:
1) up to 64 zones per substation.  The hw building block is an 8 zone (i2c
comm with 24vac opto-isolated from 5vdc) valve sensor board
2) up 32 sensors per substation. (sensors include leak detection, temp
sensors, and other stuff coming).
3) The ability to have multiple substations integrated and presented as a
single UI
4) text/email events driven by substation availability, and sensor triggers
etc.
5) program operation driven off of time or sensor triggers
6) the ability to have substations that are connected via either wifi or RF
radio so that wifi is not required everywhere
7) the abiity to have remote sensors that are radio connected to a
substation.  Remote radios can be AC or DC powered.  Eventually I will have
the radio device control dc valves in addition to just sensors.
8) evapotranspiration or sensor driven based valve control
9) simple camera integration to see that what you expect is what you are
getting

The radio infrastructure is a bit pricey but pretty cool.  I think I can
provide the valve/sensor boards at a reasonable price, but more expensive
than rays for the foreseeable future.

One of my friendly users is trying to distribute temperature sensors (via
radios) in his orchard so if it gets too cold at night he can turn on wind
machines to prevent the young fruit from freezing.  For now we are just
trying to monitor temperatures and send alerts but eventually we may turn
on the wind machines and he can see with the camera that the wind machine
actually started.

-brian

Tony-in-Oz, 2016-03-21:
Thanks Dan and bkoblenz 
Actually I found and read up on Matias's fork last night and saw that he has plugins for various GPIO expansions.  
Thanks both for your swift replies
Tony

Tony-in-Oz, 2016-04-15:
Hi again,
don't know if its just me or myself...
having trouble getting the MCP23017 working with SIP.
I have installed the Adafruit drivers.
The MCP chip is connected at 0x20 and tests ok.  And I've tried several SIP versions - but none actuate the ports on the MCP chip.  
I'm using the Relay Board plugin as I can't see any way of actually selecting them over the normal GPIO pins.  
Am I simply missing the point or just missing a step?

jbeech3, 2016-04-15:
I just figured this out in the last couple of days.  I am also using the
mcp23017 with a 16 port relay board. I just had to write a new GPIO pins
Python file to do my thing.  Don't have access to my pi right now but if
you want I can email you my file and you can see what I did or just use it
as is.  Was quite happy with my self as I had not done much in Python b4

Tony-in-Oz, 2016-04-15:
That would be great thanks

jbeech3, 2016-04-15:
The following is my gpio_pins.py file.  The tweaks I made start at line 213: that was the only module needing a change and I just added the new part.  This lets you use all 16 ports.  Still use the actual PI pin GPIO 8 for your rain sensor (rain sensor goes between GPIO 8 and Ground).  In options, Station set it to use “1” extension board to get access to all 16 zones on the MCP23017.

 

Let me know if you need help on which wires to connect between PI and MCP23017, or questions on wiring the relay.

 

Good luck.

 

Jerry Beech

 

 

    # ************************* Added by JDB 4-2016

    # this is the new version for using with the MCP23017 i2c shift register:

    rz = ''  ## Setup variable to old the value of currently running zone(s)

    rbin  = '11111111'  ### First 8 zones on A side of shift register ### All zones OFF

    rbinB = '11111111'  ### Second 8 zones on B side of shift register ## All zones OFF

 

    import smbus

    import gv

 

    bus    = smbus.SMBus(1) # Rev 2 Pi uses 1

    DEVICE = 0x20 # Device address (A0-A2)

    IODIRA = 0x00 # Pin direction register for Bank A

    IODIRB = 0x01 # Pin direction register for Bank B  

    OLATA  = 0x14 # Register for outputs for Bank A

    OLATB  = 0x15 # Register for outputs for Bank B

    GPIOA  = 0x12 # Register for inputs for Bank A

    GPIOB  = 0x13 # Register for inputs for Bank A

 

   # Set all pins to outputs  [[[ NOTE THIS MUST be CHANGED for rain sensor, etc]]]

    bus.write_byte_data(DEVICE,IODIRA,0x00)  ## All pins assumed to be outputs

    bus.write_byte_data(DEVICE,IODIRB,0x00)  ## All pins assumed to be outputs

  

 

    # loop through zones and set string rbin to be a binnary string showing which zones are on

    for num in range(len(gv.srvals)):        # Cycle thrpough zones to see if it is on 

 

               if gv.srvals[num] == 1:  # This zone number "num" is ON

                    rz  = rz + str(num+1)

                    if num < 8:      # if first 8 zones

                        rbin = rbin [:num]  + '0' + rbin [num+1:]

                    else:        # if zones 9-16

                                             rbinB = rbinB [:num] + '1' + rbinB [num+1:]

                              else:                    # Zone is OFF

                    if num < 8:  # if first 8 zones

                                             rbin = rbin [:num]  + '1' + rbin [num+1:]

                                  else:        # if zones 9-16

                                             rbinB = rbinB [:num] + '1' + rbinB [num+1:]

 

# Reverese the binnary string so zone 1 is 11111110  and zone 8 is 01111111

    print (rbinB)

    rbin  = rbin  [::-1] 

    rbinB = rbinB [::-1]

 

    # Create a integer of the decimal value of the rbin string

    runzonedecimal  = int(rbin,2)

    runzonedecimalB = int(rbinB,2)

    print 'Number of stations shown on system: ' + str(gv.sd['nst']) 

    print 'Currently running zones: ' + rz 

    print 'Rbin for A and B:' + rbin  + ' ' + rbinB 

    print 'Running zone decimal values A and b: ' +str(runzonedecimal) + ' ' + str(runzonedecimalB) 

    print 'Value of variable gv.srvals:  ' + str(gv.srvals) 

 

    # write to the device with the active zones to activate / deactivate zones

    bus.write_byte_data(DEVICE,OLATA,runzonedecimal)    

    bus.write_byte_data(DEVICE,OLATB,runzonedecimalB) 

   

    print ('.......New Zone Running Cycle Completed in gpio_pins.py set_output ........\n')

 

# **********************************  End - Added by JDB 4-2016

Dan-in-CA, 2016-04-15:
Jerry,

Thanks  for your input on this.

I think the code you posted here might be a bit hard to use because of the formatting in the post. 
Could you email a copy of your modified gpio_pins.py file to:
dkimberling59@gmail.com
and I will post it as a Gist here for easy download.

Dan

jbeech3, 2016-04-15:
Here you go.

 

 

# !/usr/bin/python
# -*- coding: utf-8 -*-

import gv

try:
    import RPi.GPIO as GPIO
    gv.platform = 'pi'
    rev = GPIO.RPI_REVISION
    if rev == 1:
        # map 26 physical pins (1based) with 0 for pins that do not have a gpio number
        if gv.use_pigpio:
            gv.pin_map = [0,0,0,0,0,1,0,4,14,0,15,17,18,21,0,22,23,0,24,10,0,9,25,11,8,0,7]
        else:
            gv.pin_map = [0,0,0,0,0,5,0,7,8,0,10,11,12,13,0,15,16,0,18,19,0,21,22,23,24,0,26]
    elif rev == 2:
        # map 26 physical pins (1based) with 0 for pins that do not have a gpio number
        if gv.use_pigpio:
            gv.pin_map = [0,0,0,2,0,3,0,4,14,0,15,17,18,27,0,22,23,0,24,10,0,9,25,11,8,0,7]
        else:
            gv.pin_map = [0,0,0,0,0,5,0,7,8,0,10,11,12,13,0,15,16,0,18,19,0,21,22,23,24,0,26]
    elif rev == 3:
        # map 40 physical pins (1based) with 0 for pins that do not have a gpio number
        if gv.use_pigpio:
            gv.pin_map = [0,0,0,2,0,3,0,4,14,0,15,17,18,27,0,22,23,0,24,10,0,9,25,11,8,0,7,0,0,5,0,6,12,13,0,19,16,26,20,0,21]
        else:  # ***** Used by jdb RPI2
            gv.pin_map = [0,0,0,3,0,5,0,7,8,0,10,11,12,13,0,15,16,0,18,19,0,21,22,23,24,0,26,0,0,29,0,31,32,33,0,35,36,37,38,0,40]
    else:
        print 'Unknown pi pin revision.  Using pin mapping for rev 3'
    
except ImportError:
    try:
        import Adafruit_BBIO.GPIO as GPIO  # Required for accessing General Purpose Input Output pins on Beagle Bone Black
        gv.pin_map = [None]*11 # map only the pins we are using
        gv.pin_map.extend(['P9_'+str(i) for i in range(11,17)])
        gv.platform = 'bo'
    except ImportError:
        gv.pin_map = [i for i in range(27)] # assume 26 pins all mapped.  Maybe we should not assume anything, but...
        gv.platform = ''  # if no platform, allows program to still run.
        print 'No GPIO module was loaded from GPIO Pins module'

from blinker import signal
zone_change = signal('zone_change')

try:
    if gv.use_pigpio:
        import pigpio
        pi = pigpio.pi()
    else:
        GPIO.setwarnings(False)
        GPIO.setmode(GPIO.BOARD) #IO channels are identified by header connector pin numbers. Pin numbers are 
except Exception:
    pass

global pin_rain_sense
global pin_relay

try:
    if gv.platform == 'pi':  # If this will run on Raspberry Pi:
        GPIO.setmode(GPIO.BOARD)
        pin_rain_sense = gv.pin_map[8]
        pin_relay = gv.pin_map[10]
    elif gv.platform == 'bo':  # If this will run on Beagle Bone Black:
        pin_rain_sense = gv.pin_map[15]
[... 175 lines cut ...]
                    if num < 8:  # if first 8 zones
			rbin = rbin [:num]  + '1' + rbin [num+1:]
		    else:        # if zones 9-16
			rbinB = rbinB [:num] + '1' + rbinB [num+1:]
# Reverese the binnary string so zone 1 is 11111110  and zone 8 is 01111111
    print (rbinB)
    rbin  = rbin  [::-1] 
    rbinB = rbinB [::-1]

    # Create a integer of the decimal value of the rbin string
    runzonedecimal  = int(rbin,2)
    runzonedecimalB = int(rbinB,2)
    print 'Number of stations shown on system: ' + str(gv.sd['nst']) 
    print 'Currently running zones: ' + rz 
    print 'Rbin for A and B:' + rbin  + ' ' + rbinB 
    print 'Running zone decimal values A and b: ' +str(runzonedecimal) + ' ' + str(runzonedecimalB) 
    print 'Value of variable gv.srvals:  ' + str(gv.srvals) 

    # write to the device with the active zones to activate / deactivate zones
    bus.write_byte_data(DEVICE,OLATA,runzonedecimal)	
    bus.write_byte_data(DEVICE,OLATB,runzonedecimalB)	
   
    print ('.......New Zone Running Cycle Completed in gpio_pins.py set_output ........\n')

# **********************************  End - Added by JDB 4-2016

Dan-in-CA, 2016-04-15:
Cool, Thanks.
It can now be cloned or downloaded by accessing the Gist button at the very top of this page.

Tony-in-Oz, 2016-04-22:
Hi guys,
sorry for long delay before replying and thanks Jerry!
It works!  well with a minor change or so. Tested outputs on both channels ok.

Without re-posting all the code, here are my changes.  I have posted the edits on gist as revision 2.

I use a relay board needing active low to energize the relays, hence the option to set 1 or 0 to activate the outputs.
I also saw a weird habit of  'rbinB'  showing1111111111111111 (16) outputs.  This seems to be caused by not resetting 'num' for the second channel.  I don't know if it actually effected the output switching.  But it still works with my change.

# in the initial declarations, I added a variable OutputMode used latter
    GPIOA  = 0x12 # Register for inputs for Bank A
    GPIOB  = 0x13 # Register for inputs for Bank A
    OutputMode = 1 # set 1 for relay requires low, or 0 to turn on

# made use of that variable and fixed a display error for channel b.  It would start showing 16 outputs, not 8!
        	if gv.srvals[num] == OutputMode:  
                    # This zone number "num" is ON - 1 = on
                    rz  = rz + ' ' + str(num+1)
                    if num < 8:	 # if first 8 zones
                        rbin  = rbin  [:num] + '0' + rbin  [num+1:]
                    else:        # if zones 9-16
			rbinB = rbinB [:num-8] + '0' + rbinB [num+1:]

		else:                    # Zone is OFF, 0 = off
                    # rz  = rz + ' ' + str(num+1)
                    if num < 8:  # if first 8 zones
			rbin  = rbin  [:num] + '1' + rbin  [num+1:]
		    else:        # if zones 9-16
			rbinB = rbinB [:num-8] + '1' + rbinB [num+1:]
                        # Tony # num-8 stops channel b from showing 16 outputs

closed 2016-08-19 by Dan-in-CA