Dan-in-CA/SIP #146: Manage Plugins option duplicated & not working

roydonm, 2016-03-31:
I updated SIP earlier today (I cannot remember what version I was on before, but it was branded as OpenSprinkler, so I think it was about 1 year old) and since then the Plugin's menu is not working.  The 'Manage Plugins' option appears twice (with System Update below that) and both the Manage Plugins links give an error (The error page simply says 'internal server error').

Dan-in-CA, 2016-03-31:
This can happen if the software does not re-start properly after the update.

Try rebooting the Pi and let me know what happens.

roydonm, 2016-03-31:
Good man !  A reboot solved it.  Thanks so much for a really great project, I've been using this for a year and it's flawless !

roydonm, 2016-03-31:
I think I spoke a bit early.  The 'manage plugins' works, but I have the 'Relay Board' plugin enabled and when I click the 'Relay Board' link from the plugins dropdown menu, I get that same error page.  Is there supposed to be something behind that link, or is it purely to control the output pins with no actual GUI behind it ?

Dan-in-CA, 2016-03-31:
There should be a configuration page linked from the plugins dropdown.

try adding /rb to the URL of your Pi. That should open the page. If not try re-installing the plugin.

roydonm, 2016-03-31:
It took a few installs/uninstalls with a few reboots, but in the end it worked.  Also had to reset the number of relays and rename all the stations, but ultimately everything is back to where it should be.  Thanks again Dan !

closed 2016-03-31 by roydonm

Dan-in-CA, 2016-03-31:
Glad you got things working. 
Sorry for the trouble.. Unfortunately sometimes software changes cause unintended problems.

elongo, 2017-05-15:
Hi Dan,

I just way in the SIP interface, PLUGIN page that there is a new version available:
"New version is available!
Available revision: 3.2.5 (2017-05-13)
Changes:
  60377dd Add station run completed blinker signal"
  
- Tried to update the system but got the page "internal server error" as well. 
- Even though the update didn't seem to go through, when running sip.py I am getting this message:
python sip.py
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Traceback (most recent call last):
  File "sip.py", line 17, in <module>
    import gv
  File "/home/pi/SIP/gv.py", line 106, in <module>
    from helpers import load_programs, station_names
  File "/home/pi/SIP/helpers.py", line 27, in <module>
    from gpio_pins import GPIO, pin_rain_sense, pin_relay
  File "/home/pi/SIP/gpio_pins.py", line 71, in <module>
    pi.set_mode(pin_rain_sense, pigpio.INPUT)
  File "/usr/lib/python2.7/dist-packages/pigpio.py", line 1291, in set_mode
    return _u2i(_pigpio_command(self.sl, _PI_CMD_MODES, gpio, mode))
  File "/usr/lib/python2.7/dist-packages/pigpio.py", line 977, in _pigpio_command
    sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
AttributeError: 'NoneType' object has no attribute 'send'

- Rebooted the RasPi and the issue remains. 

(As you may remember I having also troubleshooting the start of the valves, and thought this update would help somehow, but I am experiencing these issues now. I'll keep you posted on any progress and solutions found)

Thanks in advance.

e