Dan-in-CA/SIP #149: Stations do not work after update

stephenyo, 2016-05-05:
After updating to 3.1.46 yesterday on my rPi model B (withOSPi board), my stations no longer work.  The log shows that each station is opening correctly, but there is no water coming out.  I'm fairly new to SIP- is there another log I can check?

Dan-in-CA, 2016-05-05:
There is only one log file.
Have you tried running a station manually or from the Run Once page?

stephenyo, 2016-05-05:
Yeah, I should have mentioned I tried both of those methods as well

Dan-in-CA, 2016-05-05:
I updated an old version of OSPi here and it seem to be working properly.

A few questions to try and find a solution:
How did you do the updater? From the command line or using the system updater plugin?
Did you update from a version of OSPi or an older version of SIP? From you original post I'm guessing it was from OSPi.

If you use the command line I would suggest re-naming the OSPi directory then installing a fresh copy of SIP. See the instructions on the Wiki:
https://github.com/Dan-in-CA/SIP/wiki

Then you can copy your settings from the old OSPi directory.

Let me know if you need more details.

stephenyo, 2016-05-05:
Thanks- I did it via command line.  I'll follow your directions and report back.

Dan-in-CA, 2016-05-05:
The settings files are in the Data directory. If you copy the files
programs.json
sd.json
snames.json or snames.txt 
from the old data dir to the new one you should have all your settings intact.

Also you should follow the directions for starting the program automatically - the recommended method.
Be sure to disable the old ospi startup:
sudo update-rc.d ospi remove

and delete the ospi file in /etc/init.d/
sudo rm /etc/init.d/ospi

stephenyo, 2016-05-06:
I went ahead and did a full wipe and reinstall of the Pi as well as SIP.
I'm good to go now - thanks!

Dan-in-CA, 2016-05-06:
Cool. That seems like the best solution.

closed 2016-05-06 by Dan-in-CA