Dan-in-CA/SIP #151: 16x2 LCD Plugin

gsg2016, 2016-06-06:
fantastic software Thankyou
Having a problem getting the LCD to work.
wired the screen as per Rototron instructions PCF8574 i2c
I get a flashing courser thats it 
Got the pylcd2.py from Rimco/OSPy

Dan-in-CA, 2016-06-06:
Rimco's fork is a different code base from SIP. The plugins from there probably won't work.

you could try the " lcd_adj" plugin from the SIP plugins repo (Plugins menu > manage plugins > Browse more plugins) but the author of that plugin has not been active for some time.

martinpihrt, 2016-07-12:
Hi,
I originally was dedicated project ospi (now SIP)  plug-ins, then I started to redo all project ospy (rímco branch), where there is already plenty of other my plugins. If you want fans can try this branch. Thank you Martin. LCD plugin in OSPy is full function
PS: I thank Dan for the work he has done

https://github.com/martinpihrt/OSPy
and more plugins for OSPy
https://github.com/martinpihrt/OSPy-plugins-temp
and 
https://github.com/martinpihrt/OSPy-plugins-core

web: https://pihrt.com/elektronika/248-moje-rapsberry-pi-zavlazovani-zahrady

closed 2016-08-20 by Dan-in-CA