Dan-in-CA/SIP #152: Program Setup

slukaskawcez, 2016-06-07:
Not sure how to setup the following without having to take into account the weather adjust %.

Zone 1 -> Run for 10 minutes
Zone 2 -> Run for 15 minutes
Zone 3 -> Run for 20 minutes
Zone 4 -> Run for 20 minutes
Zone 5 -> Run for 25 minutes
Zone 6 -> Run for 5 minutes.

In the current programming scheme I would have to run Zone 3 and 4 at 6:00am then create a program to run Zone 5 at 7:20am (80 minutes later taking into account the 200% weather adjust possibility)

Is there an easier way?

Thanks,
Stacy

Dan-in-CA, 2016-06-07:
That would be best if you need to run all the stations on the same days. 
Having the plugin increase the run time instead of decreasing it is the problem. If the increased run time goes passed the start time of the next station that next station will be skipped.

slukaskawcez, 2016-06-08:
Thanks for the info, I sort of figured it was working that way.

closed 2016-08-20 by Dan-in-CA