Dan-in-CA/SIP #154: log

mrbeezer, 2016-06-28:
getting an internal server error when trying access the log, which is enabled,  Everything else seems ok.

Dan-in-CA, 2016-06-28:
The log file may be corrupt. 
Try clicking the CLEAR LOG button at the bottom of the log page. That will create a new log file.

begeles2000, 2016-06-28:
the SD card might be corrupt also, mine started failing this way and then
started to get worse

mrbeezer, 2016-06-28:
Thanjs for responding but the screen I get is all white with the error
message in black at the top.
 
I renamed the SIP dir and reinstalled and it is working now..  Love this
program! Thanks!  Is there a plugin that aloows writing the date to a usb HD
or network drive?
 
Thanks again!
 

  _____  

Zeks5qQX08gaJpZM4JAU4R> the thread.
<https://github.com/notifications/beacon/ATQZ4jhhmmq1wlEELn7bXNnKj8DfXH6Zks5
qQX08gaJpZM4JAU4R.gif>

Dan-in-CA, 2016-06-28:
Not a plugin but there are instructions on the SIP wiki that tell how to do it:
https://github.com/Dan-in-CA/SIP/wiki/flashdrive

mrbeezer, 2016-06-29:
Thank you sir!  I have been in hi-tech a long time (40 yrs) and these new
innovations are wonderful.  Keep it up!

closed 2016-08-19 by Dan-in-CA