Dan-in-CA/SIP #156: Installation Error

sbenaloh, 2016-08-05:
I just flashed a card with a new image of Raspian, cloned this repository, and when I went to run ospi.py, I got this error message:

> 
> %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
> Can't connect to pigpio at localhost(8888)
> 
> Did you start the pigpio daemon? E.g. sudo pigpiod
> 
> Did you specify the correct Pi host/port in the environment
> variables PIGPIO_ADDR/PIGPIO_PORT?
> E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888
> 
> Did you specify the correct Pi host/port in the
> pigpio.pi() function? E.g. pigpio.pi('soft', 8888))
> %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
> Traceback (most recent call last):
>   File "sip.py", line 16, in <module>
>     import gv
>   File "/home/pi/SIP/gv.py", line 48, in <module>
>     from helpers import password_salt, password_hash, load_programs, station_names
>   File "/home/pi/SIP/helpers.py", line 26, in <module>
>     from gpio_pins import GPIO, pin_rain_sense, pin_relay
>   File "/home/pi/SIP/gpio_pins.py", line 71, in <module>
>     pi.set_mode(pin_rain_sense, pigpio.INPUT)
>   File "/usr/lib/python2.7/dist-packages/pigpio.py", line 1053, in set_mode
>     return _u2i(_pigpio_command(self.sl, _PI_CMD_MODES, gpio, mode))
>   File "/usr/lib/python2.7/dist-packages/pigpio.py", line 848, in _pigpio_command
>     sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
> AttributeError: 'NoneType' object has no attribute 'send'
>

Dan-in-CA, 2016-08-06:
The program should automatically detect if pigpio is installed and use the default gpio library.

Are you using the latest version of raspbian jessie?

closed 2016-08-20 by Dan-in-CA

arlinjv, 2016-09-12:
Hi. I have the exact same problem. I do have the latest version of Raspbian running.

Dan-in-CA, 2016-09-12:
I just cloned SIP and didn't see the error.
Could you try cloning SIP again?

arlinjv, 2016-09-13:
Thanks for getting back so quick. I deleted the directory, stopped the service (per your instructions in the wiki) and ran 'sudo python sip.py' and got the same as sbenaloh (pasted below).

Then as suggested in the output I ran 'sudo pigpiod' and then 'sudo python sip.py' and it worked. I'd like to be able to run this continuously as a service. Should sip.sh be updated in some way? 

%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888))
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Traceback (most recent call last):
  File "sip.py", line 17, in <module>
    import gv
  File "/home/pi/SIP/gv.py", line 48, in <module>
    from helpers import password_salt, password_hash, load_programs, station_names
  File "/home/pi/SIP/helpers.py", line 27, in <module>
    from gpio_pins import GPIO, pin_rain_sense, pin_relay
  File "/home/pi/SIP/gpio_pins.py", line 71, in <module>
    pi.set_mode(pin_rain_sense, pigpio.INPUT)
  File "/usr/lib/python2.7/dist-packages/pigpio.py", line 1053, in set_mode
    return _u2i(_pigpio_command(self.sl, _PI_CMD_MODES, gpio, mode))
  File "/usr/lib/python2.7/dist-packages/pigpio.py", line 848, in _pigpio_command
    sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
AttributeError: 'NoneType' object has no attribute 'send'

Dan-in-CA, 2016-09-13:
Actually if you are running Raspbian Jessie (the latest version) we should be using systemd to start SIP as a service rather than init.d (sip.sh). The instructions on the SIP wiki have not been updated yet but here are the instructions for starting SIP as a service using systemd:

If you are already running sip using init.d (sip.sh):
disable with:
sudo update-rc.d sip remove
remove the old init.d file:
sudo rm /etc/init.d/sip

Install the systemd unit file (from the SIP directory):
sudo cp sip.service /lib/systemd/system/

Enable:
sudo systemctl enable sip.service

sudo reboot

systemd commands:

sudo:
systemctl status sip
systemctl stop sip
systemctl start sip
systemctl restart sip

When SIP is running as a service and you issue "sudo systemctl status sip" it will show any error messages.

###################

The pigpio Python module should be imported by SIP at startup (gpio_pins.py line ~46) .  It sounds like you have that part working.

I guess it might be necessary to make a pigpio systemd unit file to start the pigpio daemon. Not everyone uses the pigpio library so this has not been a problem.

For now you might try starting the pigpiod deamon using rc.local:
`sudo nano /etc/rc.local`
Add the command pibpiod before the exit 0 line. There are instructions for editing rc.local on the SIP wiki.

Dan-in-CA, 2016-09-13:
I just had another idea.
It should be possible to issue a command to start the pigpiod deamon from inside SIP. 
I will look into it.

Dan

arlinjv, 2016-09-13:
Thanks! I really appreciate your effort on this. I'll give it a go this weekend.

arlinjv, 2016-09-13:
I found some time to give it a try - I did have to add the line to rc.local but now it's good to go. Thanks again!

Dan-in-CA, 2016-09-13:
I edited the previous comments. Pigpio should start without needing the rc.local line.

Dan

arlinjv, 2016-09-16:
Awesome, I'll give that a try.