Dan-in-CA/SIP #157: No SIP service running

sbenaloh, 2016-08-05:
I got SIP up and running automatically when the RPi boots up (I used an sd card with the original OSPi firmware instead of pure Raspian), and I built a plugin.  To debug, I'm supposed to stop the sip service and start it manually, but when I execute "sudo service --status-all", I don't see any sip service, even though the web app is running fine and producing outputs on the board.  I do see an "OpenSprinkler.sh" service that I had to stop when I installed this firmware, but nothing else relating to SIP or OSPi.

Dan-in-CA, 2016-08-06:
Are you sure the program is not started by /etc/rc.local ? That would not show a service running. 
If so you would need to comment out the start command in rc.local and reboot the Pi.

Dan-in-CA, 2016-08-20:
After 2 weeks I am guessing this is resolved.

closed 2016-08-20 by Dan-in-CA