Dan-in-CA/SIP #160: Error when seeing stations

hespanha, 2016-08-22:
I have just installed version: 3.1.53 (2016-08-19) but when I press the "stations" button I get "internal server error". This did not use to happen.

Dan-in-CA, 2016-08-22:
I tested that version here and I am not seeing the problem.
Did you upgrade from an older version or was it a new install? Are there any settings that might affect the station names like the language setting on your system?

hespanha, 2016-08-22:
Hi Dan,

I upgraded from an older version using the command line (git pull). In the meanwhile, I did a clean install and copied the files in the old data folder to the one. I still go an error, but after a reboot it seems to work.
Joao

Dan-in-CA, 2016-08-22:
It sounds like you have fixed the problem.
When SIP is running as a service it continues to run with the old code even if you change the SIP files. That is probably why you still got the error.
When you did the reboot it started running the new code.

hespanha, 2016-08-22:
Thanks for getting back to me. Yes, all looks well. I’ll let you know if something comes up.
Joao

closed 2016-08-22 by Dan-in-CA