Dan-in-CA/SIP #161: Station Specific Water Level Adjustments.

nathantsmith, 2016-10-08:
I am picking up and old project using some solar powered sensors  to measure water on a zone by zone basis. They send back moisture and temperature data using sub-ghz radio's to  data to a hub attached to custom sip hardware and I like to rewrite the zone by zone water level control.   I've come to realize that there is no native way to set the water level on a zone by zone basis. If there was a zone by zone water level adjustment parameter developers could use the SIP framework to  not only control how much water they're yard/produce/etc  receives in general but be have the ability to control the water distribution to specific zones as they often use water up at different rates. Data analytics could tell users how moisture levels are through the day and automated water level control would allow users to water to a desired level and save extra water.

begeles2000, 2016-10-08:
Im also trying to do the same, i have a prototype sensor in process of
beeing built.
Maybe we could share some info or try to do a standard easy to build sensor
that people can build easily.

Dan-in-CA, 2016-10-08:
Hi Nathan,
That is a very interesting comment. 
I agree that zone-by-zone adjustment would be a desirable feature. Let me give this some thought. 
I am abut to begin some work on SIP to make it's web UI mobile responsive. and now would be a good time to add new features. There will also be a change to the format of program data which may facilitate on the fly changes to irrigation times.

I will probably have some questions. Is there a way I can contact you?

Dan-in-CA, 2016-10-08:
@Nathan,
What sort of signal would your system provide to SIP to set a zone's duration?
It is currently possible using a signal already implemented to know when a station/zone is started and which station is currently running. It should be fairly easy to adjust the stop time for that station.

Dan-in-CA, 2016-10-08:
@begeles2000,

I would be very interested in learning more about your sensor. Currently I have been using vegetronix sensors:
http://www.vegetronix.com/Products/VH400
and using a hard wired connection over standard sprinkler wire.

nathantsmith, 2016-10-08:
You can all contact me via email at Nathan.taylor.smith@gmail.com

nathantsmith, 2016-10-08:
@Dan-in-CA,

I was hoping I could modify a station specific global variable through some api. I'll look through the code this week maybe I'll have some ideas after. 

@begeles2000,

Would love to talk more, maybe we can work together.

nathantsmith, 2016-10-08:
I'm using a watermark moisture sensor,
 http://www.irrometer.com/sensors.html
And a semtech sx1272 for wireless communication. The watermark requires a special circuit to measure as you need to read resistance without using a D.C. Current.

Dan-in-CA, 2016-10-09:
Nathan,

The global var that holds the stop time for the currently running zone is"gv.rs". See the file gv_reference.txt in the SIP directory.

There are some http.get commands that can control SIP. You can down a PDF listing them from the SIP documentation wiki:
https://github.com/Dan-in-CA/SIP/wiki/SIP_GET_commands.pdf

They do not include changing the gv.rs var but that could be easily added with a plugin.

The code that checks the stop time is in sip.py ~line 91

Dan-in-CA, 2019-10-26:
No activity on this thread in a long time. Closing.

Re-open if there is still interest.

closed 2019-10-26 by Dan-in-CA