Dan-in-CA/SIP #165: gv.os_cname errors

Pelado-Mat, 2016-11-22:
In both of my rpi with debian jessie, gv.os_cname returns error.

Perhaps the call to lsb-relese should be changed to use the python [platform module](https://docs.python.org/2/library/platform.html)

Dan-in-CA, 2016-11-22:
What error are you getting?

The code that calls "lsb_release -c" is in a try-except block so it should return "not found" if there is an error.

This code was added as part of an attempt to automate the installation of the init script for new SIP installations. It can be commented out for now.

Pelado-Mat, 2016-11-23:
the error is catched by the try-except block. In some Jessie images lsb_release is not installed.

Dan-in-CA, 2016-11-23:
Ah, OK. 
That is good to know. 
I am taking that code out for now. It should have been in a separate branch.

closed 2016-11-23 by Dan-in-CA