Dan-in-CA/SIP #166: Error when updating SIP

ndesimone, 2016-11-30:
Configuration:
- Software version 3.1.53
- One AC powered OS Zone Expander
- Only default plugins installed (system_update, mobile_app, signaling_examples)

Problem:
When using the the System Update plugin to attempt to upgrade my system to current (3.1.77), I get an "Internal Server Error".  All other functionality seems to work, EXCEPT every time I go to the plugin section (manage or system update), I get an "Internal Server Error".  To clear the error I need to restart the SIP service ("sudo service sip restart").

I know I can re-sync from the repo directly, but I thought I'd report this to help others in the community.  Are there any logs I can provide to help with this?

Dan-in-CA, 2016-11-30:
Thanks for the heads up.
I'll take a look.

Dan-in-CA, 2016-11-30:
I pushed a fix for the system_update plugin.
The only problem is that you will need to update SIP to get the fix for the updater. It should avoid problems in the future though.

closed 2016-11-30 by Dan-in-CA

ndesimone, 2016-12-09:
Sorry for the delay (the Git notifications were going to an old work address).  

Thanks for jumping on this so quickly.  I'll give it a try.

ndesimone, 2016-12-09:
Hey Dan, 
I tried updating the system_update plugin (looked successful, but I can't actually tell).  Then tried to update SIP, but I still got the "Internal Server Error".  Got other weird behaviors like multiple System Updates appearing in the dropdown.  "Internal Server Error" when navigating to Manage Plugins.

[image]

ndesimone, 2016-12-09:
Restarting SIP got system back in order.  

Do you want me to keep testing solutions, or just upgrade manually?