Dan-in-CA/SIP #167: Negative numbers in Master On Delay crash SIP

Pelado-Mat, 2016-12-05:
The tool tips inform the user that the numbers mus be between 0 and +60, but it allows entering negative numbers in "Master on adjust"

If you, incorrecly, input a negative value SIP will crash when a program is run:
 File "/opt/SIP/sip.py", line 84, in timing_loop
 schedule_stations(p[7:7 + gv.sd['nbrd']])  # turns on gv.sd['bsy']
`
The user must be warned and we should not save an invalid configuration in sd.json

Dan-in-CA, 2016-12-05:
Got it, Thanks. I'll work on fixing this. 
We really need a way to start the master before other stations start so the master can e.g. start a pump.

Foe now I will try to prevent the error.

Dan-in-CA, 2016-12-05:
I pushed a fix.
Let me know if you still see any problem.

closed 2016-12-05 by Dan-in-CA