Dan-in-CA/SIP #170: Error when updating SIP

ndesimone, 2016-12-15:
Dan,
Sorry, I added some info/screenshots to #166 after you closed it (I wasn't paying attention).  I'm still having a problem updating.  I ran SIP from the command line and got:

`
pi@ospi ~/SIP $ sudo python sip.py
plugins loaded:
  mobile_app
  plugin_manager
  signaling_examples
  system_update
zones changed
 Starting timing loop

[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
http://0.0.0.0:8080/
Fetching origin
192.168.123.20:63909 - - [15/Dec/2016 07:23:14] "HTTP/1.1 GET /UPs" - 200 OK
192.168.123.20:63909 - - [15/Dec/2016 07:23:14] "HTTP/1.1 GET /api/log" - 200 OK
Already on 'master'
warning: CRLF will be replaced by LF in helpers.py.
The file will have its original line endings in your working directory.
warning: CRLF will be replaced by LF in plugins/mobile_app.py.
The file will have its original line endings in your working directory.
warning: CRLF will be replaced by LF in plugins/proto.py.
The file will have its original line endings in your working directory.
warning: CRLF will be replaced by LF in plugins/signaling_examples.py.
The file will have its original line endings in your working directory.
warning: CRLF will be replaced by LF in plugins/system_update.py.
The file will have its original line endings in your working directory.
warning: CRLF will be replaced by LF in webpages.py.
The file will have its original line endings in your working directory.
warning: CRLF will be replaced by LF in helpers.py.
The file will have its original line endings in your working directory.
warning: CRLF will be replaced by LF in plugins/mobile_app.py.
The file will have its original line endings in your working directory.
warning: CRLF will be replaced by LF in plugins/proto.py.
The file will have its original line endings in your working directory.
warning: CRLF will be replaced by LF in plugins/signaling_examples.py.
The file will have its original line endings in your working directory.
warning: CRLF will be replaced by LF in plugins/system_update.py.
The file will have its original line endings in your working directory.
warning: CRLF will be replaced by LF in webpages.py.
The file will have its original line endings in your working directory.
error: Your local changes to the following files would be overwritten by merge:
	helpers.py
	plugins/signaling_examples.py
	plugins/system_update.py
	webpages.py
Please, commit your changes or stash them before you can merge.
Aborting
Traceback (most recent call last):
  File "/home/pi/SIP/web/application.py", line 239, in process
    return self.handle()
  File "/home/pi/SIP/web/application.py", line 230, in handle
    return self._delegate(fn, self.fvars, args)
  File "/home/pi/SIP/web/application.py", line 423, in _delegate
    return handle_class(cls)
  File "/home/pi/SIP/web/application.py", line 399, in handle_class
    return tocall(*args)
  File "/home/pi/SIP/plugins/system_update.py", line 143, in GET
    perform_update()
  File "/home/pi/SIP/plugins/system_update.py", line 119, in perform_update
    output = subprocess.check_output(command.split())
  File "/usr/lib/python2.7/subprocess.py", line 544, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
CalledProcessError: Command '['git', 'merge', '-X', 'theirs', 'origin/master']' returned non-zero exit status 1

192.168.123.20:63909 - - [15/Dec/2016 07:23:27] "HTTP/1.1 GET /UPu" - 500 Internal Server Error
`

Interestingly, it's saying that I have local changes to helpers.py, plugins/signaling_examples.py, plugins/system_update.py, webpages.py.  I'm not why there would be any local changes to any files on this SIP install.

I'm sure I can upgrade manually, but wanted to vet this for others.  Please let me know if I should just upgrade manually and be done, or you want to keep working on this.

Thanks,
Nick

Dan-in-CA, 2016-12-15:
Hi Nick,
I ran into the same problem.
The best thing to do is update manually. 
I will do some more testing.

ndesimone, 2016-12-15:
Ok, thanks.

ndesimone, 2016-12-15:
Looks fine after manual update.  The real test will be after the next commit.

Dan-in-CA, 2016-12-15:
Yeah. The next commit is coming shortly. At least a merge of a pull request.
Steve Blake has added the ability to set individual duration times for each station in a single irrigation program. So far it has not had any problems.

ndesimone, 2016-12-15:
Ok, I'll be your tester for the upgrade process now that my system is clean.

ndesimone, 2016-12-16:
The upgrade part worked fine, but I'm now getting internal server error when trying to edit a program:

plugins` loaded:
  mobile_app
  plugin_manager
  signaling_examples
  system_update
zones changed
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
 Starting timing loop

http://0.0.0.0:8080/
Traceback (most recent call last):
  File "/home/pi/SIP/web/application.py", line 239, in process
    return self.handle()
  File "/home/pi/SIP/web/application.py", line 230, in handle
    return self._delegate(fn, self.fvars, args)
  File "/home/pi/SIP/web/application.py", line 423, in _delegate
    return handle_class(cls)
  File "/home/pi/SIP/web/application.py", line 399, in handle_class
    return tocall(*args)
  File "/home/pi/SIP/webpages.py", line 399, in GET
    return template_render.modify(pid, prog)
  File "/home/pi/SIP/web/template.py", line 1020, in template
    return self._base(t(*a, **kw))
  File "/home/pi/SIP/web/template.py", line 881, in __call__
    return BaseTemplate.__call__(self, *a, **kw)
  File "/home/pi/SIP/web/template.py", line 808, in __call__
    return self.t(*a, **kw)
  File "templates/modify.html", line 34, in __template__
    stationsOn = 0
AttributeError: 'int' object has no attribute 'extend'

192.168.123.20:50959 - - [16/Dec/2016 13:47:36] "HTTP/1.1 GET /mp" - 500 Internal Server Error

Dan-in-CA, 2016-12-16:
Yes, I just updated my production unit and I am getting the same error.
It seems to be because the upgrade is trying to extend a value that needs to be converted to an array.

I'll let you know when I find a fix.

Dan-in-CA, 2016-12-17:
OK I pushed a fix. A work around that adds an empty list to the program data on load.

Dan-in-CA, 2016-12-17:
There are still bugs. 
The new individual station duration code breaks if more stations are added.
Looks like this will take some time to fix.

ndesimone, 2016-12-17:
Hey Dan, that didn't fix it for me.  The upgrade looked good:

http://0.0.0.0:8080/
Fetching origin
remote: Counting objects: 3, done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 3 (delta 0), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (3/3), done.
From https://github.com/Dan-in-CA/OSPi
   9bce1c4..855625b  master     -> origin/master
192.168.123.20:59824 - - [17/Dec/2016 06:42:42] "HTTP/1.1 GET /UPs" - 200 OK
192.168.123.20:59824 - - [17/Dec/2016 06:42:43] "HTTP/1.1 GET /api/log" - 200 OK
Already on 'master'
rm: cannot remove `sessions/*': No such file or directory
192.168.123.20:59824 - - [17/Dec/2016 06:42:55] "HTTP/1.1 GET /UPu" - 303 See Other
System is restarting
zones changed
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
192.168.123.20:59824 - - [17/Dec/2016 06:42:55] "HTTP/1.1 GET /restart" - 200 OK
192.168.123.20:59824 - - [17/Dec/2016 06:42:56] "HTTP/1.1 GET /api/log" - 200 OK
Restarting...

However, when editing any program, I get:

ALARM from Example_Sender!: Just and example!
plugins loaded:
  mobile_app
  plugin_manager
  signaling_examples
  system_update
zones changed
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
 Starting timing loop

http://0.0.0.0:8080/
192.168.123.20:59869 - - [17/Dec/2016 06:44:45] "HTTP/1.1 GET /vp" - 200 OK
192.168.123.20:59869 - - [17/Dec/2016 06:44:46] "HTTP/1.1 GET /api/log" - 200 OK
Traceback (most recent call last):
  File "/home/pi/SIP/web/application.py", line 239, in process
    return self.handle()
  File "/home/pi/SIP/web/application.py", line 230, in handle
    return self._delegate(fn, self.fvars, args)
  File "/home/pi/SIP/web/application.py", line 423, in _delegate
    return handle_class(cls)
  File "/home/pi/SIP/web/application.py", line 399, in handle_class
    return tocall(*args)
  File "/home/pi/SIP/webpages.py", line 399, in GET
    return template_render.modify(pid, prog)
  File "/home/pi/SIP/web/template.py", line 1020, in template
    return self._base(t(*a, **kw))
  File "/home/pi/SIP/web/template.py", line 881, in __call__
    return BaseTemplate.__call__(self, *a, **kw)
  File "/home/pi/SIP/web/template.py", line 808, in __call__
    return self.t(*a, **kw)
  File "templates/modify.html", line 34, in __template__
    stationsOn = 0
AttributeError: 'int' object has no attribute 'extend'

192.168.123.20:59869 - - [17/Dec/2016 06:44:51] "HTTP/1.1 GET /mp" - 500 Internal Server Error
Traceback (most recent call last):
  File "/home/pi/SIP/web/application.py", line 239, in process
    return self.handle()
  File "/home/pi/SIP/web/application.py", line 230, in handle
    return self._delegate(fn, self.fvars, args)
  File "/home/pi/SIP/web/application.py", line 423, in _delegate
    return handle_class(cls)
  File "/home/pi/SIP/web/application.py", line 399, in handle_class
    return tocall(*args)
  File "/home/pi/SIP/webpages.py", line 399, in GET
    return template_render.modify(pid, prog)
  File "/home/pi/SIP/web/template.py", line 1020, in template
    return self._base(t(*a, **kw))
  File "/home/pi/SIP/web/template.py", line 881, in __call__
    return BaseTemplate.__call__(self, *a, **kw)
  File "/home/pi/SIP/web/template.py", line 808, in __call__
    return self.t(*a, **kw)
  File "templates/modify.html", line 34, in __template__
    stationsOn = 0
AttributeError: 'int' object has no attribute 'extend'

192.168.123.20:59869 - - [17/Dec/2016 06:44:59] "HTTP/1.1 GET /mp" - 500 Internal Server Error

Dan-in-CA, 2016-12-17:
Getting closer to a proper fix. 
Still having a problem with the new feature causing a hang when the number of stations in increased by changing the number of expansion boards under Options > Station handling.

ndesimone, 2016-12-18:
Thanks for all your hard work and attention.  Please let me know how I can help.

Dan-in-CA, 2016-12-18:
Thanks for your interest in the SIP project. 
It is community involvement that makes it so enjoyable to work on.

I think I finally fixed the bugs in the new individual station duration feature. Check it out and let me know if you find any problems. It should be enabled by default. If not you can enable it under Options > Station Handling.

Most of the bugs occurred when the number of stations was changed by increasing or decreasing the number of expansion boards.

ndesimone, 2016-12-19:
That could be it.  I have an expansion board attached to my OSPi.  I'll give it a test.

ndesimone, 2016-12-19:
Bummer didn't quite work.  I:

- Upgraded SIP (clean, no errors)
- "Individual Duration" was not enabled, so I enabled it
- Got internal error when editing a program (list index out of range)

192.168.123.20:64894 - - [18/Dec/2016 20:46:32] "HTTP/1.1 GET /api/log" - 200 OK
Traceback (most recent call last):
  File "/home/pi/SIP/web/application.py", line 239, in process
    return self.handle()
  File "/home/pi/SIP/web/application.py", line 230, in handle
    return self._delegate(fn, self.fvars, args)
  File "/home/pi/SIP/web/application.py", line 423, in _delegate
    return handle_class(cls)
  File "/home/pi/SIP/web/application.py", line 399, in handle_class
    return tocall(*args)
  File "/home/pi/SIP/webpages.py", line 399, in GET
    return template_render.modify(pid, prog)
  File "/home/pi/SIP/web/template.py", line 1020, in template
    return self._base(t(*a, **kw))
  File "/home/pi/SIP/web/template.py", line 881, in __call__
    return BaseTemplate.__call__(self, *a, **kw)
  File "/home/pi/SIP/web/template.py", line 808, in __call__
    return self.t(*a, **kw)
  File "templates/modify.html", line 409, in __template__
    $ status = ("stationOn" if stationEnabled else "") + (" dayOn" if program[1]&(1<<d) else "")
IndexError: list index out of range

192.168.123.20:64894 - - [18/Dec/2016 20:46:37] "HTTP/1.1 GET /mp" - 500 Internal Server Error

Dan-in-CA, 2016-12-19:
Hmm..
I set up my test rig with the same configuration as yours with the individual duration feature off. 
Then enabled individual durations, edited the run times and saved the changes without an error.

Did you get the error when you tried to save the program? Also, were you editing an existing program or adding a new one? I tried both without a problem.

I appreciate your continued feedback on this.

Dan-in-CA, 2016-12-19:
I also made a change to the system_update plugin. I wonder if your update actually took.

ndesimone, 2016-12-19:
SIP System Update is showing that the system is up to date (we figured that).  "git status" is saying that the working directly is clean.

BTW, the error occurred when I clicked to edit an existing program.

I'll fiddle a little and post anything interesting.

Glad to help,
Nick

ndesimone, 2016-12-19:
- Editing a new program (with individual durations turned on) works without error
- Enabling/disabling any program works without error
- Editing any existing programs causes and error

I'm wondering if this is really due to the individual duration feature.  None of my programs should be recurring (all are weekly).  Looking at the programs below, they all look recurring.  I wonder if something changed there?

[image]

I can't really edit the old programs to test my theory, but I guess I could delete them and recreate them.

ndesimone, 2016-12-19:
Or, notice that the other programs say "with individual station durations".  These were created be for the individual station feature.  They probably don't contain the data to support the feature.  Where do the programs get converted from old style to new style?

Dan-in-CA, 2016-12-19:
The programs are stored in SIP/data/programs.json and can be edited there.

The conversion from old to new version happens mostly in SIP/templates/modify.html, lines 12 - 36 and  lines 42 - 43

Also in SIP/templates/programs.html lines 131 - 132

ndesimone, 2016-12-20:
Hey Dan, sorry for the delay.  Been a bit busy getting ready for the holidays and doing some job searching.  I'll spend some time this afternoon looking at the data and conversion code.

ndesimone, 2016-12-20:
Problem is definitely related to the format change.  If I manually edit the file and add the station duration array to the end of the program, it can be successfully edited in the UI.  Right now, I suspect the problem is in the conversion from old format to new format.

Dan-in-CA, 2016-12-21:
It has to do with the fact that when more stations (expansion boards) are added, an integer is added at what was the end of the program array. The new duration array is at the end of the program array so that any new station ints need to be inserted just before the duration array. That is what this code does:

		if not isinstance(program[bid+7], int):
			program.insert(-1, 0)

I still have not been able to replicate your error.

ndesimone, 2016-12-21:
In my case, I've had the expansion board since the beginning.  So, it wasn't added later.  When I look at my programs (I'll attach a copy - that should help).  The integers are there, but the duration array isn't.  I suspect the problem is in the line 23-37 area of modify.html.  This code looks like it tries to read whatever format programs you have and convert them to what's needed.

Dan-in-CA, 2016-12-21:
I loaded your programs.json file on my test rig.
The only error I see is when I try to save a program e.g. program #2 that does not have druations for all the active stations. In that program stations 9, 13, 15 have durations but 10, 14, 16 do not. That will bring up a dialog that says that says "Error: incorrect duration" when the save button is clicked. That dialog was included with the contributed code and probably should say something like "Some durations are missing".

Are you getting some other error?

ndesimone, 2016-12-21:
BTW, to manually fix a program, I just copied the sduration array to the end of a non-editable program to make it editable.  

I just saw your reply.  Try editing program 1.  Program 2 is the one I manually fixed and the last program is the new one that worked perfectly.

Dan-in-CA, 2016-12-21:
I can edit program #1 without an error even if I delete the duration array from the programs.json file.

Dan-in-CA, 2016-12-21:
Just pushed an update with a better error message.
Try updating your copy and see if anything changes.

ndesimone, 2016-12-21:
The error and stack are exactly the same.

Couple things: 
- Looking at the error  

    $ status = ("stationOn" if stationEnabled else "") + (" dayOn" if program[1]&(1<<d) else "")
IndexError: list index out of range

The only index I see is program[1].  This seems weird to not exist, so my guess is the entire program array must be toast by this time.

- I'm confused about the code:

`	else: #  An existing program
		program = eval(prog)
		sduration = []		
		predicted_len = 7 + gv.sd['nbrd'] + 1 #  Expected length of program including sduration list
		if len(program) < predicted_len:
		    if isinstance(program[-1], list): # If program has sduration list at the end...
		        sd_list = program.pop()
		        predicted_len -= 1
		    else:
		        sd_list = [0] * 8 * (predicted_len - len(program)) # Handle older program format (no existing sduration) 
			for brd in range(predicted_len - len(program)): 
				program.extend([0]) # add bytes per number of extra boards.
				sduration.extend([0] * 8) # increase length of sduration
			program.append(sd_list)	
		if isinstance(program[-1], list):
			program[-1].extend(sduration)
		else:
			program.append([0] * 8 * gv.sd['nbrd']) # Handle older program format
`

> If we pop() the sduration array off the back of the JSON at line 7, where does it get added back?
> sd_list, if it's even needed, should be initialized outside of the if statement since is used in both the if and else conditions

Dan-in-CA, 2016-12-21:
Popping the durations list was an easy way to be able to add more station bytes if needed. 
It gets added back at line 32:
program.append(sd_list)

ndesimone, 2016-12-21:
Please bear with me, Python isn't my usual language.  

For sd_list to be available, doesn't it need to be declared above line 24?  

For sure this isn't contributing to my problem, but thought I should raise it nonetheless.

ndesimone, 2016-12-21:
A couple more thoughts:
- In the future, instead of using bitmapped station bytes, why not just use station duration > 0
- When converting from old style to new style programs, why not seed the sduration array with the old station time value?

Dan-in-CA, 2016-12-21:
In Python you can create a var by just assigning to it.

When SIP is moved to the new program data format a lot of this will go away. The program array will be replaced with a Python dictionary (associative array)

ndesimone, 2016-12-21:
That part I realized, it was just a question of scope for me.  I just did a test and Python doesn't seem to confine the scope the same way as other languages do.  All good.

ndesimone, 2016-12-22:
Hmmm.  The whole reason for digging into the code I was digging into was the error message:

$ status = ("stationOn" if stationEnabled else "") + (" dayOn" if program[1]&(1<<d) else "")
IndexError: list index out of range

I assumed something was happening to the program and program[1] was somehow invalid.  So I changed the code to:

$ status = ("stationOn" if stationEnabled else "") + (" dayOn" )

but I still get:

  File "templates/modify.html", line 409, in __template__
    $ status = ("stationOn" if stationEnabled else "") + (" dayOn" )
IndexError: list index out of range

I'm going to delete, recreate my programs.  If the problem goes away, I'll close this and move on.

ndesimone, 2016-12-22:
Yep.  Deleting all the old programs fixed it.

closed 2016-12-22 by ndesimone

Dan-in-CA, 2016-12-22:
OK, Cool.

Now I just need to write some documentation for the SIP wiki to cover the new feature.

myspaghetti, 2019-11-28:
Another issue: imported `programs.json` progs may be set to recurring even when they're single-pass.

`programData.json` does not have a boolean "recurring" switch, but the old `programs.json` does. Old programs may have `['cycle_min']` set when they are being imported even if they are not recurring, which makes them recurring.

I'm guessing the change to the new `programData.json` occurred over three years ago, so this bug probably doesn't affect anyone except people who haven't updated in three years.

Dan-in-CA, 2019-11-28:
The new programs format is new along with the Python3 compatibility.

There is a utility program "convert_progs.py" which needed some work because it was developed before the option for individual station run times was added.
It is being updated/debugged and should be posted after some additional testing, probably tomorrow. The old version of the programs data is left intact as "programs.json" the new format is stored in the file "programData.json" both in the SIP/data directory.

As soon as the conversion program is working properly you can delete the "programData.json" file and ether restart sip or run the conversion program from the command line in the SIP directory with:
`python convert_progs.py`
to generate a new programData.json file.

for now you may need to do some manual tweaking of the irrigation schedules. In some cases it may just require re-saving the program.

Dan-in-CA, 2019-11-28:
I just posted an updated (not fully tested) copy of the convert_progs.py file to GitHub.
You can do a git pull, delete the programData.json file and restart SIP. That should hopefully git your irrigation programs working.

Dan-in-CA, 2019-11-28:
The old programs.json format evolved from some Arduino code and had a bunch of bitwise logic between different bytes. Here are some of my notes about interval programs:

['day_mask']      v[1] =	1, //days byte (includes a flag for odd - even restriction) Mon. = 1, high bit: 0 = no restriction and no interval, 1 = odd/even parity or interval enabled.
//When used for interval: high bit = interval on, rest of byte used for days countdown set as "starting in" = days remaining in cycle.

The new format is much easier to grasp.

myspaghetti, 2019-11-28:
I honestly did not expect a fix! Yup, the new format is better, dictionaries are better than bitwise lists. Thanks again for this incredibly useful program.

Dan-in-CA, 2019-11-28:
I just pushed a new update for the convert_progs.py program. It fixes a problem with interval programs.
You can just run
`python convert_progs.py`
from the SIP directory and it will overwrite the "programData.json" with the correct schedules from an existing "programs.json" file.

I expect to merge the Py3_WIP branch into the main branch today or tomorrow.