Dan-in-CA/SIP #172: Please add support for H3 Orange Pi boards

marksev1, 2016-12-16:
Please add support for them...They are really cheap and good boards, there is a WiringPI fork WiringOPI which works nicely (with python bindings also)..And there is also another library which works good with them.
Or if you can't do it, can you maybe guide me how I could port those libraries to this project.
Thank you!

Dan-in-CA, 2016-12-16:
This is the first I've heard of those boards. Looks very interesting.
It should be pretty easy to get SIP running on them if there is a library similar to RIP.GPIO or if the pigpio library will run on it:
http://abyz.co.uk/rpi/pigpio/download.html

If you can get pigpio to run on your board you should be able to run SIP.
Let me know how it goes. I will try to help you if there are problems.

Dan

Dan-in-CA, 2016-12-16:
Checking a bit more it looks like pigpio might not work:
https://github.com/joan2937/pigpio/issues/86

marksev1, 2016-12-17:
There is this: https://github.com/duxingkei33/orangepi_PC_gpio_pyH3 and this: https://gist.github.com/lanefu/f16a67195c9fa35c466c6b50cdaeadea

Dan-in-CA, 2016-12-17:
Looks like
https://github.com/duxingkei33/orangepi_PC_gpio_pyH3
would probably be the best choice.

Have you tried installing SIP on your board? I think it should run even without access to the GPIO pins.
See the installation instructions on the SIP wiki:
https://github.com/Dan-in-CA/SIP/wiki/Installation

Are you up for a little code hacking? It shouldn't be to complicated. Also, do you plan to use a relay board to interface SIP to your sprinkler system?

After you try installing the software we can go from there.

mihugo, 2016-12-17:
What I did when running the software on a pcduino (an older board I had around the house)  was to get a compatible IO library working and in the python path.  In my case I just create a library called RPi and stuck it in the root directory of where the rest of the SIP software is.  No other code modification were necessary since  I used  the same IO name.  A quick glance appears that the lib Dan is suggestion might just work fine for you as well  without any code changes.

There is  a second issue that  the temperature sensor may not give the right temp on the home page.  That will require some code changes to make work.  It is only cosmetic to leave broken.
That code is in helpers.py and will take a bit of research on  your board to find the how to get the  temperature.  In my case I had to do some bit banging of some registers.  Here is the code section 
  elif gv.platform == 'pi':
            command = "cat /sys/class/thermal/thermal_zone0/temp"
        

Dan: I've thought about trying to produce a patch so that this is temp is read from a lib or some sort of plugin but I can't even win the argument with myself on how best to do it.  Do you have some thoughts on how to make this portable between hardware versions without having to hardcode each type of hardware?

marksev1, 2016-12-17:
Im up for it but gonna need to wait for my second board to arrive from China. Because one of them i have in "production" and cant do dev work on it :). Thanks for the willingness to help! Yeah I think I plan to use a diy relay board or something like that.

Dan-in-CA, 2016-12-17:
For a relay board you might want to Google something like the Keyes 5V relay boards. They are inexpensive and ready to use.
You might find this interesting (OSPi is the old name for SIP):
https://github.com/KanyonKris/relay_board/wiki

An alternate way to connect:
https://github.com/Dan-in-CA/SIP/wiki/Relay-interface

Dan-in-CA, 2016-12-17:
@mihugo,
The temperature sensor issue is and interesting one. It assumes that all embedded CPUs have a built in temp sensor.
There is some code in SIP to try and figure out what platform it is running on. Perhaps the temperature code could be put into importable modules (python files) and the required one imported at startup.

marksev1, 2017-01-24:
Got another question, so I'm gonna need how many pins to control these relay boards? Will they be digital pins?

Dan-in-CA, 2017-01-24:
The relays use digital pins.
The number of pins used depends on how you connect the relay boards. If you only need a few stations (relays), you would connect each relay directly to a pin. If you will need more stations you can use a shift register interface to connect the relays and only use a few pins. This is described on the SIP wiki:
https://github.com/Dan-in-CA/SIP/wiki/Relay-interface

marksev1, 2017-01-28:
Ok so I'm gonna try and start to prepare stuff since after the Chinese new year I will finally order the boards :)...so you guys said i should use the pyA20 library https://github.com/duxingkei33/orangepi_PC_gpio_pyH3 ...Can you give me some quick pointers on what i should do :)

Dan-in-CA, 2017-05-23:
No activity for nearly 4 months. Closing for now.

closed 2017-05-23 by Dan-in-CA

marksev1, 2017-05-23:
Yeah sorry, got to many stuff I have to do. Will ping you when I start working on it :).

Dan-in-CA, 2017-05-23:
OK. Sounds good.

I know how it is when things get busy.

ScreamZ, 2023-04-17:
Any news 5 years later ? :D