Dan-in-CA/SIP #173: Feature Suggestion:  Display Total Run Time

ndesimone, 2016-12-19:
Background:
I have several programs that I use for different areas of the garden (beds vs. lawn vs. garden)

Request:
Would like the program tab to display total runtime for each program so that I can:
- See how much water is to be used by each program
- Know when a program should be complete so and can schedule the next one to start

Note: This may be less of a priority once the "individual station duration" feature is fully shaken out.

Dan-in-CA, 2016-12-19:
I guess you mean a total run time for the individual station durations in a program? The old version did show the run time for each program.

That should be fairly easy to implement. The individual run times for each program are stored as a list of durations in seconds. Python has a "sum" function that works on lists and the result can be converted to hh:mm:ss for display.

Currently the data stored in the programs.json file are in a sort of convoluted format. The plan is to convert the data to a more straight forward Python dictionary format which will make it much easier to work with. Much of the work has already been done and there is a branch named "prog_restruct" which contains the changes. I have been holding off on implementing the change because it will break compatibility with the OpenSprinkler mobile app which a lot of folks use with SIP. The individual station duration  feature also adds a new wrinkle. Now that SIP can be used directly on mobile devices such as smart phones I plan to move to the new format soon but it will need more work and testing.

At this point I would like to minimize changes to the core program until the program format is changed. The individual station duration contribution was too good to pass up.

ndesimone, 2016-12-19:
No problem, as I mentioned, the individual station duration capabilities may allow me to get rid the multiple programs.

ndesimone, 2016-12-20:
After spending time on #170, I see what you mean by the data format.

Dan-in-CA, 2019-12-30:
Cleaning up old issues.

closed 2019-12-30 by Dan-in-CA