Dan-in-CA/SIP #175: Feature request - custom station order within sequential program

embak, 2017-01-27:
To avoid water hammer problems and control watering sequence, my current SIP setup consist in a bunch of single station programs with manually adjusted the start time + duration to get a few seconds overlap.
With the new individual duration feature added to the sequential program and a negative value for station delay, the only missing degree of freedom is the possibility to reorder stations within a program.

Any plan to add user defined station order capability to sequential programs ?

Dan-in-CA, 2017-01-27:
That is an interesting request and I have given it some thought.

SIP uses ordered lists to store information such as station names, outputs to valves, start times, durations, etc. When a program starts to run there is a list that contains durations for the stations that are to run in that program. SPI steps through the list and if there is no duration at a location in the list it skips to the next location. When a duration value is found, SIP uses the location of that value in the list to  associate the duraton with the corresponding station. Because of this SIP always runs the stations in the same order.   

I don't see a practical way to change the order of stations within a program. However it might be possible to create a plugin that could change the run order. It could be more complicated to set the run order for each program using a plugin than it is to make separate programs as you have done.

If water hammer is a problem there are inexpensive devices called water hammer arresters  that attach to the water system and absorb the water hammer's pressure spike sort of like a shock absorber on a car.
For example:
http://www.homedepot.com/s/water+hammer+arrester?NCNI-5

embak, 2017-01-27:
Regarding the water hammer problem, I already tried such arresters without
any success.

Another motivaton for this feature will be to allow the combination of
program sequence + water hammer protection (using negative value of station
delay) + ordered sequencing per program of the stations (ex: start by the
garden, followed by the flower beds and finish with the lawn).

I have 22 active zones.  It will also be interesting to adjust the watering
duration based on humidity sensors (through a plugin).  Tring to adjust 22
individual programs start time and duration so they still overlab by a few
seconds is not really practical and maintanable.

I briefly looked at the code and it is possible to do that.  The major
impact will be on the program configuration file structure, the WEB UI and
the sequencer.

Any hint or interest on your side.

Dan-in-CA, 2017-01-27:
Yes! I would certainly be interested if you would like to give it a go.

embak, 2017-07-27:
Just to give-you an update. 
I made the modifications to the timing loop + web interface + simulation part and it works pretty well.
I am now addressing the web program edition part (modify.html).
The modifications will also correct a few potential bugs and streamline the way gv.rs[] and gv.ps[] are manipulated, plus it will allow programs to runs over midnights (this was my case).
The main drawback is that the format of the programs will be modified:
- replacing program "end time" by a program "total duration"
- removing the nbrd x active station packed values 
- adding a program "sequence duration",
- replacing the list of individual station duration at prog[-1] by a nested list of sequence composed of : station id + duration + enable + concurrent/sequential mode.
BTW I made a function that detect and convert the older formats to this new one at startup.
I know that it won't be fully compatible with the OpenSprinkler apps, but it will work on mobile using the browser.
I intend to finish-it soon.
I will let you know when it is ready.

Dan-in-CA, 2019-12-30:
Cleaning up old issues.

closed 2019-12-30 by Dan-in-CA